raspiblitz/raspiblitz #1099: Windows OpenSSH client recommendation

openoms, 2020-03-28:
I am just trying the windows openssh client in Powershell. Works really well. Easier than Putty/Kitty, might worth recommending in the readme.

https://www.howtogeek.com/336775/how-to-enable-and-use-windows-10s-built-in-ssh-commands/

[image]

openoms, 2020-03-28:
[image]

In this case only need to delete or edit the `Windows(C:)/Users/USERNAME/.ssh/known_hosts` file

closed 2020-04-18 by rootzoll

kaiobe, 2020-05-03:
I installed the Ubuntu 20.04 app from the the MS Store which uses the Windows Subsystem for Linux (WSL). Allows you to use Linux Terminal and run Linux command line utilities. Believe there's a version for a bunch of Linux distros.
https://www.microsoft.com/store/productId/9N6SVWS3RX71

openoms, 2020-05-04:
@kaiobe this is after you have activted WSL, right? https://docs.microsoft.com/en-us/windows/wsl/install-win10
It takes a couple more steps.