raspiblitz/raspiblitz #1103: Feature request: native way to migrate to a larger SSD

jrimbolt, 2020-04-01:
**Is your feature request related to a problem? Please describe.**
Many users that have started running a node in the past years will have bought a 500GB SSD, as per the Raspiblitz guide. With the blockchain size growing, and many (awesome!) services that require additional disk space (indexes) being added to Raspiblitz, SSDs are likely to start filling up in the coming year or so. This will likely cause a mass-migration to larger SSDs. I believe this process can be streamlined for users by making it easy to do natively through the Pi itself.

Currently, moving from a 500GB SSD to a larger one requires either 1) to set up a new Raspiblitz and copy the blockchain data from the old SSD to the new SSD or 2) to use a Linux partition manager on a separate system.

**Describe the solution you'd like**
Ideally, users would be able to plug in a second, larger SSD directly into the Pi and the existing Raspiblitz system on the 500GB SSD to the new larger SSD directly from the Raspiblitz menu/terminal and be able to follow a simple walkthrough of the process.

**Describe alternatives you've considered**
Another alternative would be to provide a step-by-step guide/video of how to migrate existing Raspiblitz setups to a larger SSD, as for example [this nodl video](https://www.youtube.com/watch?v=H8uq5eEcgOs).

There are probably other alternatives I'm not aware of.

rootzoll, 2020-04-01:
Since version v1.4 there is the possiblity to export/import a complete "Migration File" - especially for such situations to change the HDD/SSD or migrate to another disk format. Please see if this fixes your need: https://github.com/rootzoll/raspiblitz/blob/master/README.md#import-a-migration-file

jrimbolt, 2020-04-02:
This is almost what I had in mind, but requires the blockchain data to be stored on a separate system (which usually can't be copied from the old SSD due to the format / requires significant free storage space). It would be great if users could connect the new SSD to the Pi and copy the data directly from the old. This way you simply plug in the new SSD to the Pi, use the Raspiblitz menu and some commands to transfer migration data + blockchain data, and unplug the old SSD when done. Is this possible?

rootzoll, 2020-04-02:
With the export/import the idea is not to export the blockchain and reproducable indexes - just to keep it to the essential data. To get the blockchain again you have to resync or copy from another computer like your laptop.

You could try this in a manual way, but I would not recommend this - because in the past with running two HDDs/SSDs on a RaspiBlitz always had sideffects because because of insufficient power that can lead to data curruption. Yes if people take care about adding extra power thru a USB Hub for example it can work, but the devil is in the detail - tahts at least my experience from the past. So providing an import/export that can work with just one HDD/SSD connected is the more stable way to do it.

Also not in all scenarios you have two HDDs .. if you just want to change the file formatting of the one HDD you have, you would need an import/export anyway. So defaulting for this solution for migration is more universal (keep it simple, not too many different features) and the more stable way by not stressing the power situation on on the system.

openoms, 2020-04-12:
If you have two USB-SATA adapters a solution is to clone the disk to the bigger one using Linux (sufficent to boot temporarily from USB such as an Ubuntu Live USB: https://ubuntu.com/tutorials/try-ubuntu-before-you-install#1-getting-started):

* requires using a linux partition manager
* can only clone from a smaller disk to a bigger one.

* Boot a Linux Live USB on the desktop (Ubuntu or Clonezilla for example)
* connect the disks with two usb-sata adapters to two separate USB3 connectors of the desktop/laptop.
* Start the app: Disks
* Clone the HDD to the SSD
* expand the partition on the new (bigger) disk

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll