raspiblitz/raspiblitz #1104: BTCpay Server showing LN invoices as partially paid

openoms, 2020-04-02:
Linking to this issue on the BTCpay side: https://github.com/btcpayserver/btcpayserver/issues/1343#issuecomment-606533009

Seems to be a problem with connecting to LND v0.9.0 and above, waiting for an update to fix.

openoms, 2020-04-18:
Updating to LND v0.10.0 https://github.com/rootzoll/raspiblitz/issues/1116 will break the LN invoice checking in BTCpay. It will be not possible to downgrade the nodes to the last compatible (lnd 0.9.0) version from v0.10.0 as major versions have irreversible database migrations.

Our priority is to keep up with LND.

Will update BTCpay once they sort out the compatility with LND 0.9.2 and above.
Note that this bug does not affect the onchain invoices,PayJoin capability and any other function of BTCPay.

Possible solution temporarily is to connect BTCpay to an external lightning node running LND 0.9.0.

Roasbeef, 2020-04-24:
If someone has the raw listinvoices output for an invoice showing as "partial paid" we can likely get to the bottom of this pretty quickly.

rootzoll, 2020-04-24:
Seems like RockstarDev wants also to work on this this weekend:
https://twitter.com/r0ckstardev/status/1253788719800336384?s=20

openoms, 2020-04-27:
This seems to be fixed now in BTCPay and compatible with LND v0.10.0 RC4 too. 
Will need to checkout the latest master or wait for the next release: https://github.com/btcpayserver/btcpayserver/commits/master

openoms, 2020-04-27:
Can confirm that the issue is fixed when building BTCpay with the commit: 
https://github.com/btcpayserver/btcpayserver/commit/d4dbe6fe17d844caf1636d29dc7d96414f0b18c6

If needed for the we can incorporate this version to the next release.

closed 2020-04-28 by rootzoll

pretyflaco, 2020-05-05:
Can confirm that v1.5rc2 solves this. 
👌👏