raspiblitz/raspiblitz #1105: LND service is not running

7gQdm9zxtSq, 2020-04-02:
Hello,
after updating to version 1.5 the raspiblitz was trying to reboot, but because of the shimfan issue i had to pull the plug and i restartet the pi without the fan. (allready ordered heatsink-case)
After recovering and setting up a new password A the LND service and electRS are not running.
here the debug output: https://termbin.com/exdv
thanks for your help!

edgeball, 2020-05-23:
same here

I had it before and was hoping the 1.5 Upgrade would fit it

https://termbin.com/xla7

rootzoll, 2020-05-27:
@edgeball your LND errror is "panic: freepages: failed to get all reachable pages"

This was discussied here in the LND github repo:
https://github.com/lightningnetwork/lnd/issues/4102

> It sounds like the freepages space of your database got corrupted. You could try to create a copy in safe mode that ignores the freepages. I added the compactdb command to my channel tools that does exactly that: https://github.com/guggero/chantools

felivix, 2021-01-25:
Hi

Two days ago I updated my node (on a Pi3 with a HDD) to version 1.6.3, and also tried to set up a new node (on a Pi4 with an SSD, importing the lnd-rescue file), since then the LND service is no longer running on the Pi3 and never started on the Pi4. 

On both nodes I get the error message: **panic: freepages: failed to get all reachable pages** (page xxxxxx: multiple references)
See the debug output from my old node (Pi3): https://termbin.com/gcx2 

On my old node I tried to use Oliver Gugger (guggero)'s ChanTools compactdb and started with the On-Chain Recovery:

admin@RFnode:~ $ lncli create
Using the existing cipher seed mnemonic I then got the error:
**[lncli] rpc error: code = Unavailable desc = all SubConns are in TransientFailure, latest connection error: connection error: desc = "transport: Error while dialing dial tcp 127.0.0.1:10009: connect: connection refused"**

I then tried to check the lnd's log but got the error: permission denied.

And now when I restart the old node I get the error: 
****************************************
* The LND service is not running.
****************************************
If you just started some config/setup, this might be OK.

468 RESTARTS DETECTED - LND might be in a error loop
Jan 25 06:00:08 RFnode lnd[23026]: panic: freepages: failed to get all reachable pages (page 20053: multiple references)
Jan 25 06:00:08 RFnode lnd[23026]: goroutine 24 [running]:
Jan 25 06:00:08 RFnode lnd[23026]: go.etcd.io/bbolt.(*DB).freepages.func2(0x20b6700)
Jan 25 06:00:08 RFnode lnd[23026]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/db.go:1003 +0xcc
Jan 25 06:00:08 RFnode lnd[23026]: created by go.etcd.io/bbolt.(*DB).freepages
Jan 25 06:00:08 RFnode lnd[23026]:         go.etcd.io/bbolt@v1.3.5-0.20200615073812-232d8fc87f50/db.go:1001 +0x10c

@rootzoll @guggero what shall I do to get the LND service running again?

Regards
Felix

guggero, 2021-01-25:
If you only want to run the `compactdb` command, you don't need to go through the whole recovery process. Just stop `lnd` on both Pis and run `chantools compactdb` with the correct parameters (not sure how exactly the folders look on RaspiBlitz). With that you get a smaller, hopefully fixed `channel.db` that you then use to replace the old one. Then you should be able to start the node again, getting rid of the `panic: freepages: failed to get all...` error.

felivix, 2021-01-26:
Hi Oliver @guggero 
chantools compactdb worked like a charm - easy and fast! 
With the new channel.db in place the LND service started without a problem and the balance is still intact. 
Thank you for the excellent app and the good advise.

rootzoll, 2021-01-28:
Just as a side info ... there is a PR #1962 that is planned to have "chantools" included to the repair tools for next RaspiBlitz release. Can we close this issue?

felivix, 2021-01-29:
@rootzoll 
>>Can we close this issue?
From my side yes.  Chantools in the next release is a great addition.

closed 2021-01-29 by rootzoll

dravenk, 2023-05-21:
I have the same problem. After LND running for a period of time, my hard disk was offline. I copied the hard disk data and ran it again.
> panic: freepages: failed to get all reachable pages (page 0: invalid type: unknown<00>)

lnd version 0.16.2-beta commit=v0.16.2-beta