raspiblitz/raspiblitz #1106: Screen crooked after initialization.

tongokongo, 2020-04-03:
After initial block download (with fresh setup, starting from 0) the screen is showing the values with additional tabulation. No problem while interacting with a starting screen from ssh connection. Version 1.5-rc1, RPi4 4GB. There were no problems with previous versions on this hardware.

Restarting (power-off/on) fixes the problem.
[image]

rootzoll, 2020-04-14:
I have seen similar graphic errors with LCDs that were not the model that was linked in the shopping list. If its not that LCD linked in the shopping and you get this too often, please exchange it.

nyxnor, 2021-04-18:
Pls close tjis.

closed 2021-04-19 by rootzoll