raspiblitz/raspiblitz #1110: FAIL: No raspiblitz.conf found afer migration restore

pretyflaco, 2020-04-07:
While migrating from v1.2 (rpi3) to v1.5rc1, I first created a migration file on my laptop by using 
`$ ./lnd.rescue.sh backup` on v1.2
This was succesful and it created the file on my laptop
`lnd-rescue-493cc5e2f16c828bc6bc8b3a2872f689.tar.gz`

Setting up the new rp4, I chose the Migration option after first boot and went for the experimental formatting. After this was finished, I was given the following instruction:

`
*****************************
* UPLOAD THE MIGRATION FILE *
*****************************
If you have a migration file on your laptop you can now
upload it and restore on the new HDD/SSD.

ON YOUR LAPTOP open a new terminal and change into
the directory where your migration file is and
COPY, PASTE AND EXECUTE THE FOLLOWING COMMAND:
scp -r ./raspiblitz-*.tar.gz admin@192.168.178.22:/mnt/hdd/temp

Use password 'raspiblitz' to authenticate file transfer.
PRESS ENTER when upload is done.
`
After renaming the file from lnd-rescue-* to raspiblitz-*, the Upload was found, but raspiblitz.conf was missing in the migration file:

`OK: Upload found in /mnt/hdd/temp - restoring data ... (please wait)
/home/admin/config.scripts/blitz.migration.sh: line 360: /mnt/hdd/raspiblitz.con                               f: No such file or directory

FAIL: No raspiblitz.conf found afer migration restore
`
Any idea, why the config file is missing in the migration file or how to complete migration else? 
Thx in advance!

openoms, 2020-04-07:
It is not the migration tool you need to use but the script you backed up with.

Start new, choose to set up a new node and there will be an option to upload the lnd-rescue file when getring to setup the Lightning wallet.

pretyflaco, 2020-04-07:
Ah got it, thanks. Will try and update, copying Blockchain from other node now.

pretyflaco, 2020-04-08:
For some reason, I was later not asked to add the ln-rescue file, but instead asked to unlock wallet. The info was already available and all I had to do was unlock wallet with my old wallet PW. But it worked and migration is now complete and all channels are active now.

openoms, 2020-04-08:
What you are describing is the normal SDcard update process. It works when stepping through multiple versions too:

https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#how-to-update-my-raspiblitz-since-12

closed 2020-04-09 by pretyflaco