raspiblitz/raspiblitz #1116: LND update to v0.10.0

openoms, 2020-04-15:
v0.10.0 RC1 is out. 
https://github.com/lightningnetwork/lnd/releases/tag/v0.10.0-beta.rc1

rootzoll, 2020-04-15:
OK ... this makes sense then to delay the final v1.5 release until the final LND 0.10.0 is out of the door.

openoms, 2020-04-15:
Not sure how other accompanying software like RTL, Zeus, BTCPay (still on lnd 8.2) etc will keep up. Some breaking RPC changes are said to be postponed until lnd 0.11 so the existing functionality should be kept, but needs testing.
Highlights from Optech https://bitcoinops.org/en/newsletters/2020/04/15/:
[image]

openoms, 2020-04-16:
RC2 https://github.com/lightningnetwork/lnd/releases/tag/v0.10.0-beta.rc2

openoms, 2020-04-18:
Great recap of the latest lnd developments in v0.10.0 in the video:
https://twitter.com/udiWertheimer/status/1251573484259426304?s=19

openoms, 2020-04-20:
Transcript (until the relase notes are out):
https://twitter.com/kanzure/status/1252024800974905350?s=19

openoms, 2020-04-22:
RC4, now with release notes
https://github.com/lightningnetwork/lnd/releases/tag/v0.10.0-beta.rc4

openoms, 2020-04-23:
I think it should be considered to release 1.5 with LND 0.9.2 (which has the ability even to be downgraded to 0.9.0 if needed for BTCpay for example)
0.10.0 might break some otherwise stable functions working now.
There could be an optional update fro LND only (but one way!)

frennkie, 2020-04-23:
Sound like a reasonable approach!

rootzoll, 2020-04-23:
@openoms seems OK to me - any idea when BTCPayServer will work with latest LND again?

openoms, 2020-04-23:
> @openoms seems OK to me - any idea when BTCPayServer will work with latest LND again?

The issue is linked here: https://github.com/rootzoll/raspiblitz/issues/1104#issue-592956438
BTCPay is still bundled with lnd 0.8.2 (seem to work with 0.9.0), but the lightning integration is not their priority. 

I don't think that BTCpayserver itself should hold us back updating, but I think testing things like RTL and the Loop Service is quite important before engaging with the LND new version.

openoms, 2020-04-24:
https://github.com/lightningnetwork/lnd/releases/tag/v0.10.0-beta.rc5

rootzoll, 2020-04-26:
I started to add a new script `./config.scripts/lnd.update.sh` that will offer "interims updates". There are two modes:

a) "secure" with every RaspiBlitz release we can add a secure optional LND update where we can have one more progressed version that can be installed with all the checks of signatures. This one is for situations like now, where we have a LND update with breaking changes and that is not down-gradable. So after default setup, users can easily decide to make a secure upgrade.

b) "reckless" that will just grab the latest release from the LND releases page and offer that for install (without any further signature checks). This one is for test and development nodes that what quickly test new releases.

I already implemented the basic script (waiting for final 0.10.0 release to test the secure update). If one of the update modes is used it will write a value into the `raspiblitz.conf` called `lndInterimsUpdate` so that even when you recover with a new sd card image it will privision with the choosed update again.

TODO: Adding the interim update option to the SSH GUI menus. Most likely to the "UPDATE" section.

rootzoll, 2020-04-26:
OK - there is now a sub menu when you call "UPDATE" in main menu. There you can choose "LND" and then decide todo a "SECURE" or "RECKLESS" update.

Final testing can begin as soon as the final LND 0.10.0 is out.

openoms, 2020-04-29:
It is out on GitHub: https://github.com/lightningnetwork/lnd/releases/tag/v0.10.0-beta

openoms, 2020-04-30:
Warning about updating on the Zeus TG:
[image]

openoms, 2020-04-30:
Comes with new Loop version too: https://github.com/lightninglabs/loop/releases/tag/v0.6.0-beta

rootzoll, 2020-05-01:
Arg ... the loop version seems to be dependent of the LND update. So if I need to adapt scripts for that.

rootzoll, 2020-05-01:
OK .. also loop is now updated if LND is updated. 
Last TODO for final testing: check if verified LND update and loop update is persistent over recovery.

openoms, 2020-05-03:
RTL  v0.7.0 fails the circular rebalances on lnd 0.10 with a failure finding a path.

closed 2020-05-06 by rootzoll