raspiblitz/raspiblitz #1117: tlsextraip settings for LND

openoms, 2020-04-15:
Currently if I have there are more than 1 lines of `tlsextraip` is given in the lnd.conf there is a warning on the LCD:
```
While reading from '<string>' [line 18]: option 'tlsextraip' in section 'Application Options' already exists
```
Also the default setting `tlsextraip=0.0.0.0` is not useful. 
Any IP address or domain used to connect vis REST or GRPC needs to be added as a new 
`tlsextraip=` or `tlsextradomain=` line.

See: https://github.com/lightningnetwork/lnd/blob/master/sample-lnd.conf#L31

frennkie, 2020-04-15:
Ah.. My bad. So you would suggest to not do any configuration validation from Raspiblitz side..?

openoms, 2020-04-15:
LND would not start with an erroneous lnd.conf, so I don't see the merit of checking it again.

rootzoll, 2020-04-17:
@openoms when is that happening ... the nodes I am running all just have one tlsextraip line and all is good.

openoms, 2020-04-17:
I needed to use multiple lines due to connecting through ZeroTier and using the Tor2IP tunnel too.

frennkie, 2020-04-17:
I agree. The `tlsextraip=0.0.0.0` line makes no sense and should be removed.

frennkie, 2020-04-17:
Interessting (re)discovery.. run this on your Node: `openssl x509 -in /mnt/hdd/lnd/tls.cert -noout -text | grep -A 1 "Subject Alternative Name"`

frennkie, 2020-05-10:
I opened https://github.com/lightningnetwork/lnd/issues/4271 for the "finding" in my last comment here.

frennkie, 2020-05-23:
> when is that happening ... the nodes I am running all just have one tlsextraip line and all is good.

@rootzoll The point isn't really when this is happening.. the line makes no sense to me in any scenario. There is no possibility to connect to a RaspiBlitz using `0.0.0.0` as the destination IP address.

My suggestion would be to remove the lines that add this to `lnd.conf` from `lnd.autounlock.sh`. But I'm not sure whether we should also actively remove it from `/mnt/hdd/lnd/lnd.conf`. People who care can do this themselves. And new installations will not longer have this setting added.

Let me know if I missing something.

openoms, 2021-01-08:
​Found an interesting setting here: https://github.com/alexbosworth/run-lnd
```
#​ Delete and recreate RPC TLS certificate when details change or cert expires​
​tlsautorefresh​=1
```

rootzoll, 2021-01-10:
```
# Delete and recreate RPC TLS certificate when details change or cert expires
tlsautorefresh=1

# Do not include IPs in the RPC TLS certificate
tlsdisableautofill=1
```

@openoms To not include IPs at all could maybe help a lot to make to prevent TLS changes at all.

openoms, 2021-01-10:
@rootzoll yes we should definitely try these

openoms, 2021-01-13:
> ```
> # Delete and recreate RPC TLS certificate when details change or cert expires
> tlsautorefresh=1
> 
> # Do not include IPs in the RPC TLS certificate
> tlsdisableautofill=1
> ```
> 
> @openoms To not include IPs at all could maybe help a lot to make to prevent TLS changes at all.

Tested these with good results. Only drawback was that my LND restarted without notice after regenerating the TLS.cert and sat for waiting for the password C.
The TLS.cert was expired a couple days ago, but the refresh only happened hours after I have set the lnd.conf. Did not restart after removing an IP from `tlsextraip=` eitehr so not sure what was the trigger of the restart. Anyway it is better then having the authenticated GRPC connections stop working because of the expired cert.

Old cert:
```
admin@rpi64:~ $ openssl x509 -text -noout -in /mnt/hdd/backup_lnd/tls.cert
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            51:d7:92:0f:93:71:90:a5:d2:96:e8:ba:8f:a7:7d:cb
        Signature Algorithm: ecdsa-with-SHA256
        Issuer: O = lnd autogenerated cert, CN = raspberrypi
        Validity
            Not Before: Nov  9 19:22:23 2020 GMT
            Not After : Jan  4 19:22:23 2022 GMT
        Subject: O = lnd autogenerated cert, CN = raspberrypi
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub:
                    04:bb:ad:91:f6:73:1e:9b:f2:6b:5b:3c:bb:fb:90:
                    08:06:a5:69:56:65:1b:ea:a7:46:fb:df:20:f7:c8:
                    60:bd:d3:c5:08:bb:bd:bd:9f:c2:f9:5b:ad:e6:c6:
                    d1:de:8a:d5:77:df:83:66:c3:79:a0:1f:ce:59:34:
                    5f:74:9d:42:56
                ASN1 OID: prime256v1
                NIST CURVE: P-256
        X509v3 extensions:
            X509v3 Key Usage: critical
                Digital Signature, Key Encipherment, Certificate Sign
            X509v3 Extended Key Usage: 
                TLS Web Server Authentication
            X509v3 Basic Constraints: critical
                CA:TRUE
            X509v3 Subject Alternative Name: 
                DNS:raspberrypi, DNS:localhost, DNS:unix, DNS:unixpacket, DNS:bufconn, IP Address:127.0.0.1, IP Address:0:0:0:0:0:0:0:1, IP Address:192.168.X.XXX, IP Address:XXXX:0:0:0:XXXX:XXXX:XXXX:XXXX, IP Address:XXXX:0:0:0:XXXX:XXXX:XXXX:XXXX, IP Address:EXTERNAL_IP_set_in_lnd.conf
    Signature Algorithm: ecdsa-with-SHA256
         30:45:02:20:47:9d:fe:49:b5:de:97:7d:5f:19:8b:21:0e:0d:
         a7:b2:4a:40:e8:f4:17:5a:a4:0b:ff:d2:11:56:00:b6:f4:91:
         02:21:00:b3:b6:aa:93:14:ab:6a:93:0e:9d:6d:59:09:e4:28:
         09:92:7c:9d:e2:dc:05:69:50:8b:3a:50:fa:1b:4f:bd:67
```

Also solves the issue raised by @frennkie in: https://github.com/lightningnetwork/lnd/issues/4271 by leaving out the internal IPv4 and IPv6 addresses.
```
openssl x509 -text -noout -in /home/admin/.lnd/tls.cert
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            e2:5d:07:fb:5a:1c:f7:9b:0d:83:eb:29:8d:33:ef:61
        Signature Algorithm: ecdsa-with-SHA256
        Issuer: O = lnd autogenerated cert, CN = localhost
        Validity
            Not Before: Jan 12 04:46:34 2021 GMT
            Not After : Mar  9 04:46:34 2022 GMT
        Subject: O = lnd autogenerated cert, CN = localhost
        Subject Public Key Info:
            Public Key Algorithm: id-ecPublicKey
                Public-Key: (256 bit)
                pub:
                    04:29:ce:03:3c:2a:f2:49:71:91:13:c4:e4:1a:aa:
                    65:be:90:9e:5a:63:78:f7:b5:1b:98:91:1b:22:7f:
                    a5:3d:ef:f4:d9:9b:a6:54:02:17:57:6b:ed:a3:81:
                    37:94:fb:f8:8c:15:b2:9c:38:23:f6:0b:5c:9b:ca:
                    6a:69:e8:40:8b
                ASN1 OID: prime256v1
                NIST CURVE: P-256
        X509v3 extensions:
            X509v3 Key Usage: critical
                Digital Signature, Key Encipherment, Certificate Sign
            X509v3 Extended Key Usage: 
                TLS Web Server Authentication
            X509v3 Basic Constraints: critical
                CA:TRUE
            X509v3 Subject Alternative Name: 
                DNS:localhost, DNS:unix, DNS:unixpacket, DNS:bufconn, IP Address:127.0.0.1, IP Address:0:0:0:0:0:0:0:1
    Signature Algorithm: ecdsa-with-SHA256
         30:45:02:21:00:f4:ab:12:97:68:77:92:75:b1:3a:df:bd:db:
         54:bf:04:d9:8e:eb:24:04:d4:2b:d7:7c:56:a0:51:69:4a:da:
         ff:02:20:0f:f7:14:0a:5d:b6:15:ce:89:00:17:8b:66:27:0a:
         82:f1:c1:64:68:f0:87:9a:c5:b0:0b:96:fd:41:ea:f8:55
```

We should add both flags by default to the LND.conf.

rootzoll, 2021-03-25:
OK changes were merged into dev for v1.7RC1 release for testing. But still some further questions see:
https://github.com/rootzoll/raspiblitz/pull/1943#issuecomment-807318797

rootzoll, 2021-04-16:
OK the new strategy in the config will now be added for old nodes on update/recover: https://github.com/rootzoll/raspiblitz/blob/45784b2b716c96eb1944b74fe24210f23abf77e3/home.admin/_bootstrap.migration.sh#L210