raspiblitz/raspiblitz #1120: Error while formatting 1TB SanDisk SSD Plus

bluewitch, 2020-04-16:
Having an issue with formatting the SSD, The error message instructed me to submit this to the Raspiblitz Github
[image]

rootzoll, 2020-04-17:
It looks like there is one partition on your disk that the standard procedure was not able to remove. Can you show me the output of `lsblk -o NAME,TYPE,LABEL,PTTYPE,PARTLABEL,PARTFLAGS`

cereal88, 2020-04-17:
[image]

cereal88, 2020-04-17:
ok, if you use an 32GB USB Stick u have only 29,8GB and the formatting script use the HDD/SSD to create the 30.0GB Partition. So it fails ....

openoms, 2020-04-17:
I have tested the BTRFS script and running got to work it successfully, but did not use a thumbdrive when setting up.
Needed to modify the fstab manually to fix: https://github.com/rootzoll/raspiblitz/issues/1055

cereal88, 2020-04-18:
[image]
[image]
i am confused

openoms, 2020-04-21:
put the middle 889GB partition in the fstab to be mounted in  `/mnt/storage`

Mine looks like this (redacted):
```
sudo cat /etc/fstab 
proc            /proc           proc    defaults          0       0
PARTUUID=xxxxxx-01  /boot           vfat    defaults          0       2
PARTUUID=xxxxxx-02  /               ext4    defaults,noatime  0       1
UUID=xxxxxxxxxxxxxxxx /mnt/hdd btrfs noexec,defaults,subvolid=257 0 2
UUID=xxxxxxxxxxxxxxxx /mnt/storage btrfs noexec,defaults,subvolid=257 0 2
UUID=xxxxxxxxxxxxxxxx /mnt/temp vfat noexec,defaults,uid=1002,gid=1002 0 2
```

```
$ lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0 931.5G  0 disk 
├─sda1        8:1    0    30G  0 part /mnt/hdd
├─sda2        8:2    0 867.5G  0 part /mnt/storage
└─sda3        8:3    0    34G  0 part /mnt/temp
mmcblk0     179:0    0  29.7G  0 disk 
├─mmcblk0p1 179:1    0   256M  0 part /boot
└─mmcblk0p2 179:2    0  29.5G  0 part /
```
```

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll