raspiblitz/raspiblitz #1122: Login password seems doesn't match anymore

belgacoin, 2020-04-17:
Hi, I can't login anymore into my node. Seems that the master user password doesn't match anymore. I can't imagine I have changed that.
I tried to login with different users via SSH (pi, admin, root, bitcoin). I can connect via SSH though. error states: "Permission denied, please try again".
I tried to login at the Pi itself but it doesn't work neither.
I still have the seed of course and funds are still on it (copied it on Zap wallet).
Any ideas how to solve this pls?

rootzoll, 2020-04-17:
- Get a new/second sd card.
- Put a fresh RaspiBlitz image on there.
- Shutdown the RaspiBlitz ... if you dont have touchscreen and you cannot ssh in: disconnect LAN cable, wait 5min, check that HDD or SD card LED is not blinking and pull the power
- Now replace the sd card with the new one
- StartUp again
- RaspiBlitz should do now a recovery/update process
- At the end of this you can SSH in with default password "raspiblitz" and set a new Password A

closed 2020-04-17 by rootzoll

derrickr, 2020-06-12:
Hey rootzoll,

I appreciate this is closed and my question is not mega urgent as I'm guessing you must be busy with other more pressing stuff, but I've just expereinced exactly the same issue.

I'm always Very concerned/paranoid about security and am anxious that perhaps some nefarious access has been made without my knowledge.

I've managed to follow your process above and am now back up & running, but still kinda nervous about unwanted access.

Also, just FYI, but I had to re-do my wireless config (/etc/wpa_supplicant/wpa_supplicant.conf).

Just for peace of mind, do you think my Raspiblitz could've been accessed by others? I'm running v1.5 (just upgraded to v1.5.1, due to the above) and over Tor.

Would appreciate your thoughts, but no rush.

Cheers!