raspiblitz/raspiblitz #1126: Refactor: Non-Interactive Setup Process

rootzoll, 2020-04-17:
To prepare the setup over Webbrowser in future versions, its needed to re-arrange the setup flow in that way that all setup-data (names, password, seeds, ) are gathered up-front and then are written to a setup-file and the rest of the setup runs non-interactive.

Things to keep in mind:
- Once the LND wallet was initilased to recover from SEED there should be no restart - otherwise the LND will forget the recovery window and not find any on-chain funds
- This should also be combined with a TOR as default approach

rootzoll, 2020-05-27:
Take care that also a re-factory fixes:
Make BTCPayServer install on update/recovery #1198

rootzoll, 2021-04-27:
This should be one of the firsts todo for a v1.7.1 .. because its needed as a prepartion for the v1.8 WebUI setup ans also before a possible c-lightning integration to keep merge conflicts low.

It sould be started after project planning for v1.8 WebUI is done.

rootzoll, 2021-04-28:
A more detailed description of the non-interactive setup process can be found here: https://github.com/rootzoll/raspiblitz/issues/43#issuecomment-828840191

openoms, 2021-04-29:
Would like to add some relevant notes:
* Fast update option - inactivate services before switching off for an SDcard update https://github.com/rootzoll/raspiblitz/issues/2250
* Minimize / confirm restarts after installing services: https://github.com/rootzoll/raspiblitz/issues/2249

Also we can think ahead to be able to implement full disk encryption.
In that case (this is how Nodl works) only the boot partition is unencrypted and only does offer a minimal login page to ask for the encryption password. 
Once that is given all other passwords could be used automatically (LND autounlock can be applied by default).
This would leave the user with only one password to manage and use after every restart ideally in a web interface (could be over Tor / LAN HTTPS/ LAN HTTP (fallback) or even utilise the touchscreen or a some 2FA (eg yubikey) optionally in the future).

All other passwords (for the LN wallet, bitcoinRPC, SSH login) can be randomized during the setup and shown securely over an encrypted tunnel like Tor / HTTPS / ssh.

rootzoll, 2021-04-29:
The disk encryption could be part of the boot from ssd feature #1235 - but that I see pushed back to v1.9 ... v1.8 is already a big package, we cannot do all at once.

For this issue about the "Fast update option" I dont think we need a special feature. Am I correct that your your goal is to keep Lightning downtimes minimal to not decrease QualityOfService to network? If yes then I think with regard to this issue this can be done thru:

**In the non-interactive setup/recovery process make sure that the Bitcoin & Lightning services are started as early as possible to minimize downtime.**

Additional apps can install after that - this may slow down the update/recovery process but reducing downtime to the network is getting a bigger priority now that the network matures.

openoms, 2021-04-29:
Yes, agree that building the extra services after LND is started and unlocked is the best solution.

Re. the disk encryption I wasn't thinking of implementing it just yet, but it is good to keep in mind how we could make it easier to be used in the future.

rootzoll, 2021-04-29:
This is a first draft how non-interactive provision process could be part of the future v1.8 setup:

[image]

Source File: https://docs.google.com/drawings/d/1iYYXrZTFqH5xMVxSXP7BSmf38vIj6GhXtJoVGFCHrQE/edit?usp=sharing

nyxnor, 2021-04-29:
I know the words does not fit inside the box, but focus on the arrangements.

[BlitzStructure.odf](https://github.com/rootzoll/raspiblitz/files/6402119/BlitzStructure.odf)
[image]

rootzoll, 2021-04-29:
@nyxnor sure you can see migration/setup dialogs as one part (that seither done by WebUI or SSH) but the detection if its a fresh/exiting/mograting blitz happens before the user interaction. But there is one point that the previous model maybe missed ... thats the I have a HDD from another fullnodeOS and I dont want to migrate - I just want to setup a raspiblitz and maybe just reuse the blockchain, etc.

So maybe I was missing a general step in the process, where after the `Gather All System Info` the user has to be asked what is TODO with that in detail. That would also mean that Update would not start automatically as its now. So that there always will be an interaction. Will ponder on that. I think for v1.7.1 I will keep as it is - but that can be then rearranged with the step to v1.8

But adding info on the Tor bridges can really happen so late in the process? The `Provision Process` would already has internet requests (getting binaries, etc).

nyxnor, 2021-04-29:
>So maybe I was missing a general step in the process, where after the Gather All System Info the user has to be asked what is TODO with that in detail. That would also mean that Update would not start automatically as its now. So that there always will be an interaction.

About missing general step, yeah, I understand this as a draft also, I missed some things too but that would add a lot of text and blocks for now.
The updates I imagine just clicking update from given githubUsername and that is it, no other interaction.

>But adding info on the Tor bridges can really happen so late in the process? The Provision Process would already has internet requests (getting binaries, etc).

It can be moved to the first acccess to an interface (ssh, or webui).

rootzoll, 2021-04-30:
I updated the draft .. as you can see the tor briges should be able to set in the first `System-Ready` loop:

[image]

Source File: https://docs.google.com/drawings/d/1iYYXrZTFqH5xMVxSXP7BSmf38vIj6GhXtJoVGFCHrQE/edit?usp=sharing

rootzoll, 2021-09-21:
I final setup reboot was still needed ... but basically the Setup process is now ready for the WebUI future.

closed 2021-09-21 by rootzoll