raspiblitz/raspiblitz #1128: 500GB disks filling up - delete Electrs database and txindex

openoms, 2020-04-19:
Got a first report today that an  500 GB SSD has filled up with the blockchain data, txindex (~25 GB now) and ElectRS indexing (needs ~150 GB now during indexing which is compacted down to half when finished).

Some TODO-s:
* introduce a check in the electrs setup to not start on a 500 GB disk
* add option to the `REPAIR` menu to delete the electrs index at any point and uninstall Electrs
* add option to the `REPAIR` menu to remove the txindex and dependent applications.

openoms, 2020-04-20:
Alternative to gain some time for small disks is is to implement ElectrumX: https://github.com/rootzoll/raspiblitz/issues/1130

rootzoll, 2020-04-20:
I would recommend now going with v1.5 to make the 1TB SSD the default for the standard package. Its just +50% more in price for the +100% in capacity and the prices should even fall soon even more.

rootzoll, 2020-04-20:
Also a TODO before v1.5 release: Correct estimate prices in README ... I just make the quick sum and the standard goes more for around 200 USD for all parts - not 250 USD anymore. Recheck also the other price estimates.

rootzoll, 2020-05-01:
OK I added thate REPAIR options and added a disk size check for 1TB before electrs install. This is now in final testing for v.15 release.

rootzoll, 2020-05-02:
Tested and works fine. Closing issue for release of v1.5

closed 2020-05-02 by rootzoll