raspiblitz/raspiblitz #1130: ElectrumX an Electrum Server alternative

openoms, 2020-04-20:
Advantages: 
* has a rolling compaction so takes up space only incrementally , now up to 41 GB (vs Electrs up to 125 GB during initial sync and compacting to ~ half)
* supports batch querying of addresses (dignificant when using busy wallets like with joinmarket)

Negative:  
* takes much longer to sync

Implementation notes: 
https://electrumx.readthedocs.io/en/latest/HOWTO.html#installing-on-raspberry-pi-3
https://electrumx.readthedocs.io/en/latest/HOWTO.html#running

rootzoll, 2020-04-22:
The new standard HDD/SSD will be 1TB ... see https://github.com/rootzoll/raspiblitz/issues/1128#issuecomment-616816075

So I think we dont have much pressure/prio to change the Electrum server implementation at the moment.

openoms, 2020-04-22:
ok, I am happy to close this, just wanted to make a note.
Electrs works well for personal usage, can reopen if someone would want to try ElectrumX.

openoms, 2022-02-01:
Took 6.5 days on the RPi  https://twitter.com/openoms/status/1487099572694437892 and likely superseeded by Fulcrum: https://github.com/cculianu/Fulcrum #2924

Closing again.

closed 2022-02-01 by openoms