raspiblitz/raspiblitz #1133: Bitcoin not running ssh connection always resetting after reboot.

sal-btc21, 2020-04-23:
I'm a newbie with limited experience trying to set up a node.  Can always get past set up stage including LND wallet/4 passwords etc but then after a reboot I get 'The bitcoind service is not running' and cannot establish an ssh connection.
Tried deleting keys inside ssh folder on harddrive using ubuntu which allows me ssh access again but will always lose it after reboot.  Have tried starting again from scratch... same thing.  have tried removing fan and using geekworm case..
Any ideas?  I have a bug report which I can share.
Thanks.

sal-btc21, 2020-04-23:
***************************************************************
* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.4
chainnetwork: bitcoin / main
 10:08:16 up 37 min,  2 users,  load average: 0.11, 0.33, 0.63

*** BLOCKCHAIN SYSTEMD STATUS ***
Failed to get properties: Failed to activate service 'org.freedesktop.systemd1': timed out (service_start_timeout=25000ms)

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Thu 2020-04-23 09:17:01 BST, end at Thu 2020-04-23 10:08:36 BST. --
Apr 23 09:33:01 raspberrypi systemd[1]: Starting Bitcoin daemon...
Apr 23 09:33:01 raspberrypi systemd[1]: Started Bitcoin daemon.
Apr 23 10:01:49 raspberrypi systemd[1]: bitcoind.service: Succeeded.
1587632820 ERROR: Error found in Logs
1587632891 ERROR: Error found in Logs

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-04-23T09:01:41Z UpdateTip: new best=00000000000002ea7efd29fa70c22b02d46c7aea14617363288251985ca920d8 height=206994 version=0x00000001 log2_work=68.988352 tx=8692289 date='2012-11-08T06:58:23Z' progress=0.016478 cache=336.4MiB(3012859txo)
2020-04-23T09:01:41Z UpdateTip: new best=0000000000000438c428585b7ec3d057b0b50b80551639a0b356640555a4e50d height=206995 version=0x00000001 log2_work=68.988387 tx=8692300 date='2012-11-08T07:04:21Z' progress=0.016478 cache=336.4MiB(3012865txo)
2020-04-23T09:01:42Z UpdateTip: new best=00000000000000670275e0e9c0d88a32c51b10a99b7caef783a306926420c7e0 height=206996 version=0x00000001 log2_work=68.988422 tx=8692458 date='2012-11-08T07:09:15Z' progress=0.016479 cache=336.3MiB(3012567txo)
2020-04-23T09:01:42Z UpdateTip: new best=000000000000011c1eb4c6ec30d6c4051e8b2a918cb18258e42d5a1c5a743851 height=206997 version=0x00000001 log2_work=68.988457 tx=8692546 date='2012-11-08T07:09:31Z' progress=0.016479 cache=336.3MiB(3012565txo)
2020-04-23T09:01:42Z UpdateTip: new best=00000000000001b98248a6c252ed69162801fc3fe1b81cbbbf4d3082db5d5756 height=206998 version=0x00000001 log2_work=68.988492 tx=8692574 date='2012-11-08T07:17:59Z' progress=0.016479 cache=336.3MiB(3012590txo)
2020-04-23T09:01:42Z UpdateTip: new best=000000000000009b5d4e79a6b8347be07cce3fb08f6c2e54d3bf1c5410c2f45e height=206999 version=0x00000001 log2_work=68.988527 tx=8692575 date='2012-11-08T07:28:12Z' progress=0.016479 cache=336.3MiB(3012591txo)
2020-04-23T09:01:42Z UpdateTip: new best=00000000000004bdd54a303529953b64a85eee5099acb965f53b577a1ceeca44 height=207000 version=0x00000002 log2_work=68.988562 tx=8692793 date='2012-11-08T07:28:11Z' progress=0.016479 cache=336.4MiB(3012773txo)
2020-04-23T09:01:42Z UpdateTip: new best=0000000000000478e80bae3a9f9ca2d19fe2d2e6d2f129b7fcaa804bdc17529d height=207001 version=0x00000002 log2_work=68.988597 tx=8693124 date='2012-11-08T07:35:29Z' progress=0.016480 cache=336.4MiB(3012723txo)
2020-04-23T09:01:42Z UpdateTip: new best=00000000000000f5c386ed3b4e60c03bc88d2c45201702482cfd681934e49c71 height=207002 version=0x00000001 log2_work=68.988632 tx=8693125 date='2012-11-08T07:37:05Z' progress=0.016480 cache=336.4MiB(3012724txo)
2020-04-23T09:01:42Z UpdateTip: new best=00000000000001899d88c27ff809348eaa873d1099515a0e384396e13211384d height=207003 version=0x00000001 log2_work=68.988667 tx=8693261 date='2012-11-08T07:44:21Z' progress=0.016480 cache=336.4MiB(3012768txo)
2020-04-23T09:01:42Z UpdateTip: new best=0000000000000461fa78737ce125b3aaa3715dd7b685d5e579d6e633086fdc28 height=207004 version=0x00000001 log2_work=68.988702 tx=8693343 date='2012-11-08T07:50:32Z' progress=0.016480 cache=336.4MiB(3012699txo)
2020-04-23T09:01:42Z UpdateTip: new best=0000000000000402b70233fbb8184bb5975f03e045894c0221822462887e82d7 height=207005 version=0x00000001 log2_work=68.988737 tx=8693421 date='2012-11-08T07:51:01Z' progress=0.016481 cache=336.4MiB(3012726txo)
2020-04-23T09:01:42Z UpdateTip: new best=0000000000000088f5a0208aa5e841b9a5c025935c17050c495bc093c8a42cac height=207006 version=0x00000001 log2_work=68.988772 tx=8693429 date='2012-11-08T07:52:55Z' progress=0.016481 cache=336.4MiB(3012725txo)
2020-04-23T09:01:42Z UpdateTip: new best=000000000000022198e63febbf2e93709508bc4327b5b4e82966cc8e64f9e190 height=207007 version=0x00000001 log2_work=68.988807 tx=8693582 date='2012-11-08T07:59:33Z' progress=0.016481 cache=336.4MiB(3012883txo)
2020-04-23T09:01:42Z UpdateTip: new best=00000000000001f5d10aa034a73cdcfe89e92bc858577b863a9ec1e8a16b1637 height=207008 version=0x00000001 log2_work=68.988842 tx=8693602 date='2012-11-08T08:01:27Z' progress=0.016481 cache=336.4MiB(3012884txo)
2020-04-23T09:01:42Z UpdateTip: new best=000000000000026c279d6c3abd88ee23e20f31f5d1d526ae8b9b5aae5ae442c6 height=207009 version=0x00000001 log2_work=68.988877 tx=8693758 date='2012-11-08T08:33:19Z' progress=0.016481 cache=336.4MiB(3012964txo)
2020-04-23T09:01:42Z UpdateTip: new best=000000000000026be9e801c17e9aaf7994e638ff6ae2e9608ea94d996172c4a1 height=207010 version=0x00000001 log2_work=68.988912 tx=8694083 date='2012-11-08T08:34:52Z' progress=0.016482 cache=336.4MiB(3013340txo)
2020-04-23T09:01:42Z UpdateTip: new best=00000000000004fd193c77496a1a2c951dfffe61e7bc832d154e8d6d089ef945 height=207011 version=0x00000001 log2_work=68.988947 tx=8694195 date='2012-11-08T08:42:32Z' progress=0.016482 cache=336.4MiB(3013343txo)
[... cut ...]
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=0
bitcoinErrorShort='Error found in Logs'
bitcoinErrorFull='0'
# OK: log 'ERROR: Error found in Logs' written to /home/admin/systemd.blockchain.log
startcountLightning=29
lndActive=0
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=50

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
use the following command and share the resulting link:
/home/admin/XXdebugLogs.sh | nc termbin.com 9999

rootzoll, 2020-04-23:
The " Crucial MX500 1TB" is not recommended anymore ... but normally should work, but can make problems with power stability in edge cases.

Do you have the ShimFan or already using the passive HeatSink Case running?

sal-btc21, 2020-04-24:
Using the heatsink case

sal-btc21, 2020-04-24:
Using the heatsink case. It seems that the only way to ssh in is to delete keys from ssh folder on hard drive. This gets me in one time but then if I log out after that then when I try and ssh back in I always get 'Connection reset by [my IP] port 22'

sal-btc21, 2020-04-24:
Apr 24 10:35:27 raspberrypi sshd[8796]: Accepted password for admin from [my IP] port 34952 ssh2
Apr 24 10:35:27 raspberrypi sshd[8796]: pam_unix(sshd:session): session opened for user admin by (uid=0)
Apr 24 10:56:16 raspberrypi sshd[5023]: error: Could not load host key: /etc/ssh/ssh_host_rsa_key
Apr 24 10:56:16 raspberrypi sshd[5023]: error: Could not load host key: /etc/ssh/ssh_host_ecdsa_key
Apr 24 10:56:16 raspberrypi sshd[5023]: error: Could not load host key: /etc/ssh/ssh_host_ed25519_key
Apr 24 10:56:16 raspberrypi sshd[5023]: fatal: No supported key exchange algorithms [preauth]

sal-btc21, 2020-04-27:
Anyone got any ideas?  Still stuck.

openoms, 2020-04-27:
Change that disk or add external power to it (X825 board is the only way I know works with the RPi4 for external power)
That model has froze for me and others too producing similar errors to yours:
https://github.com/rootzoll/raspiblitz/issues/905#issuecomment-581526545

sal-btc21, 2020-04-28:
Thanks.  I'll give a new SSD a try.

sal-btc21, 2020-04-30:
Result: Switched to the SANDISK SSD Plus and now it's all working fine.  Thanks

closed 2020-04-30 by sal-btc21