raspiblitz/raspiblitz #1134: ok torrented the blockchain and get.....

DrBlackross, 2020-04-23:
Inactive: Hash check on download completion found bad chunks, consider using "safe_sync".

soooo how do i set "safe_sync" and restart?

DrBlackross, 2020-04-23:
btw, update one finished no problems

DrBlackross, 2020-04-23:
FYI

*** 1) Status Torrent 'blockchain':
  [OPEN]    233736.1 / 233738.1 MB Rate:   0.0 /   0.0 KB Uploaded:  1408.5 MB                 [   R: 0.01]
  Inactive: Hash check on download completion found bad chunks, consider using "safe_sync".

*** 2) Status Torrent 'update':
Completed

rootzoll, 2020-04-23:
If one of the torrents fails you need to use "x" to abort, delete the torrent data and start again. If you have a RaspberryPi 4 and an SSD I would recommend to use the self SYNC instead the torrent anyway. Torrents are more a fallback for the slow the RaspberryPi 3 that is not able to self sync.

DrBlackross, 2020-04-24:
yea my setup is a 1.5tb ssd and rpi4... going to switch over to self-sync

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll