raspiblitz/raspiblitz #1142: [Feature Request] Balance of Satoshis and Telegram bot connection

openoms, 2020-04-27:
The Balance of Satoshis by Alex Bosworth is  feature rich tool to work with LND balances: https://github.com/alexbosworth/balanceofsatoshis
Also has an experimental feature to connect to a personal Telegram botand notify about the node activity.
* [Install instructions for the RaspiBlitz](https://gist.github.com/openoms/823f99d1ab6e1d53285e489f7ba38602)  
 
* See how to use the rebalance command with:  
    `bos help rebalance`

[image]

openoms, 2020-05-19:
Current version is shown in:
https://registry.npmjs.org/balanceofsatoshis Raw Data tab
or
https://github.com/alexbosworth/balanceofsatoshis/blob/master/package.json#L69

Once installed update with:
`sudo -u bos npm i -g balanceofsatoshis`

Can be reinstalled from the menu to reset to the tested version included in the release.

openoms, 2020-05-26:
new version 5.37.4
best to keep updated closer to the release since updated quite frequently

rootzoll, 2020-07-24:
TODO: Chech if BoS is installed correctly on upadte/recovery

rootzoll, 2020-07-24:
OK BoS now also makes a clean reinstall on recover/update. Will be part of v1.6 final. Closing issue.

closed 2020-07-24 by rootzoll