raspiblitz/raspiblitz #1143: Refactor macaroon storage and access

rootzoll, 2020-04-27:
see: https://github.com/rootzoll/raspiblitz/pull/1135#issuecomment-620123128

frennkie, 2020-04-28:
> Yes with more and more users for extra apps getting introduced, the strategy to copy them all into seperate drectories comes to a limit.

> To prevent copying we could place the original macaroons to a seperate directory in /mnt/hdd/app-data/lnd and use unix file permissions to allow different user access (lndadmin-group, lndread-group).

> @frennkie this way we could prevent any copying when macaroons get updated, correct?

Yes, I think this would be a suitable way to avoid copying around the macaroon files. Also this would make it easily checkable which user has access to which macaroon by running e.g. `id btypay` and looking at the unix group memberships.

frennkie, 2020-04-28:
I would suggest to not include "group" in the group name.. ;-)

Proposal (including IDs to make sure that they will be same on every RaspiBlitz)

GID  | Needed? | Group Name | Macaroon File Name
---- | ------- | -----------| ------------------
9700 | YES | lndadmin | admin.macaroon
9701 | ??? | lndchainnotifier | chainnotifier.macaroon  
9702 | YES | lndinvoice | invoice.macaroon  
9703 | ??? | lndinvoices | invoices.macaroon  
9704 | YES | lndreadonly | readonly.macaroon
9705 | ??? | lndrouter | router.macaroon  
9706 | ??? | lndsigner | signer.macaroon  
9707 | ??? | lndwalletkit | walletkit.macaroon

Order is alphabetical from the macaroon file as per `LND 0.9.2`.

Questions

* [ ] Should we only start with the three needed groups and add the others later is a need arises?
* [ ] I would support placing these in: `/mnt/hdd/app-data/lnd` - agreed?
* [ ] I would assume we also use the same sub directory structure: `${network}/${chain}net/` - right?

frennkie, 2020-05-23:
This is done.. right? Or do still see open tasks (e.g. changing `adminmacaroonpath` etc.)?

rootzoll, 2020-05-27:
Yeah basically this is done ... I still think having the macarrons directly written by LND to the app-data directory (instead of copy and manual keep in sync) would be best. 

Also maybe not to forget about the LNbits still - which wil be some time until that is also fixed.

So I push this to v1.7 for later impovements.