raspiblitz/raspiblitz #1146: Node syncing blockchain progress is going backwards 94.44 -> 94.43 -> 94.42

DCress12, 2020-04-28:
Hello,

New install but node syncing blockchain progress is going backwards 94.44 -> 94.43 -> 94.42

Can any one help?

rootzoll, 2020-04-28:
When it goes backwards ut can olny mean that your not is still able to get info about the growing block heights, but has somehow stopped syncing.

Can you post the termbin link to the debugLogs?
Go to console --> run `./XXdebugLogs.sh`

rootzoll, 2020-04-28:
Also sometimes a reboot helps:
Go to console --> run `menu` --> choose "OFF" -> wait until white screen and reconnect power

DCress12, 2020-04-28:
***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.4
chainnetwork: bitcoin / main
 16:57:25 up 13:09,  2 users,  load average: 0.18, 0.28, 0.29

*** BLOCKCHAIN SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
   Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-04-28 03:51:28 BST; 13h ago
  Process: 2527 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
  Process: 2532 ExecStart=/usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin/bitcoind.pid (code=exited, status=0/SUCCESS)
 Main PID: 2560 (bitcoind)
    Tasks: 18 (limit: 2200)
   Memory: 288.0M
   CGroup: /system.slice/bitcoind.service
           └─2560 /usr/local/bin/bitcoind -daemon -conf=/home/bitcoin/.bitcoin/bitcoin.conf -pid=/home/bitcoin/.bitcoin…

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST BLOCKCHAIN ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Logs begin at Tue 2020-04-28 07:20:29 BST, end at Tue 2020-04-28 16:57:25 BST. --
-- No entries --

*** LAST BLOCKCHAIN 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2020-04-28T15:57:01Z connected to self at 192.168.40.1:58528, disconnecting
2020-04-28T15:57:03Z connected to self at 192.168.40.1:65206, disconnecting
2020-04-28T15:57:03Z connected to self at 192.168.40.1:60219, disconnecting
2020-04-28T15:57:04Z connected to self at 192.168.40.1:56645, disconnecting
2020-04-28T15:57:08Z connected to self at 192.168.40.1:50116, disconnecting
2020-04-28T15:57:09Z connected to self at 192.168.40.1:53443, disconnecting
2020-04-28T15:57:10Z connected to self at 192.168.40.1:56353, disconnecting
2020-04-28T15:57:11Z connected to self at 192.168.40.1:59759, disconnecting
2020-04-28T15:57:12Z connected to self at 192.168.40.1:54830, disconnecting
2020-04-28T15:57:13Z connected to self at 192.168.40.1:51980, disconnecting
2020-04-28T15:57:14Z connected to self at 192.168.40.1:58475, disconnecting
2020-04-28T15:57:14Z connected to self at 192.168.40.1:54547, disconnecting
2020-04-28T15:57:15Z connected to self at 192.168.40.1:58184, disconnecting
2020-04-28T15:57:16Z connected to self at 192.168.40.1:56729, disconnecting
2020-04-28T15:57:17Z connected to self at 192.168.40.1:50254, disconnecting
2020-04-28T15:57:18Z connected to self at 192.168.40.1:59311, disconnecting
2020-04-28T15:57:21Z connected to self at 192.168.40.1:59874, disconnecting
2020-04-28T15:57:22Z connected to self at 192.168.40.1:53184, disconnecting
2020-04-28T15:57:23Z connected to self at 192.168.40.1:53460, disconnecting
2020-04-28T15:57:25Z connected to self at 192.168.40.1:64382, disconnecting

*** LND SYSTEMD STATUS ***
● lnd.service - LND Lightning Daemon
   Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2020-04-28 03:51:28 BST; 13h ago
  Process: 2561 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log lightning STARTED (code=exited, status=0/SUCCESS)
 Main PID: 2563 (lnd)
    Tasks: 14 (limit: 2200)
   Memory: 262.8M
   CGroup: /system.slice/lnd.service
           └─2563 /usr/local/bin/lnd --externalip=

Warning: Journal has been rotated since unit was started. Log output is incomplete or unavailable.

*** LAST LND ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Logs begin at Tue 2020-04-28 07:20:29 BST, end at Tue 2020-04-28 16:57:25 BST. --
-- No entries --

*** LAST 30 LND INFO LOGS ***
sudo tail -n 30 /mnt/hdd/lnd/logs/bitcoin/mainnet/lnd.log
2020-04-28 04:03:30.190 [DBG] CRTR: Mission control state reconstruction finished: n=0, time=26.614µs
[... 48 lines cut ...]
tempFAIL=0
tempWARN=0
tempMAX='4830 centiGrad'

*** SYSTEM STATUS (can take some seconds to gather) ***
localIP='192.168.40.100'
tempCelsius='37.5'
uptime=47376
upsStatus='OFF'
startcountBlockchain=1
bitcoinActive=1
blockchainHeight=614946
initialSync=1
syncProgress=94.38
startcountLightning=1
lndActive=1
walletLocked=0
syncedToChain=0
scanTimestamp=1588089446
scanDate='Tue 28 Apr 16:57:26 BST 2020'
scanProgress=100.00
lndRPCReady=1
blitzTUIActive=0
blitzTUIRestarts=0
scriptRuntime=1

rootzoll, 2020-04-28:
The "connected to self" in bitcoind reminded me of an older issue.
https://github.com/rootzoll/raspiblitz/issues/613

The user reported that rebooting twice solved its issue. Also check on the network connection.

rootzoll, 2020-04-28:
Oh that was you ;)

DCress12, 2020-04-28:
Yes it was. I had given up at that time and decided to give 1.4 a go.

rootzoll, 2020-04-28:
OK .. havent heard of that problem since.

we-do-it-lu, 2020-12-28:
Hi. I have the same issue, poked around on the Telegram group but no spontaneous answer there and then I found this issue still open. Is this the right channel to keep this investigation going? My node is stuck is this state,  would be happy to dig this with you

rootzoll, 2020-12-29:
@we-do-it-lu the v1.6.3 tries to fix this. if you like you can give the release candidate a test ride: https://github.com/rootzoll/raspiblitz/issues/1906

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-18 by DCress12