raspiblitz/raspiblitz #1154: Background initial synch on install

arcbtc, 2020-05-02:
Could the Blitz/bitcoind use a hosted blockchain and move the initial synch into a background operation?
So a user could start using the blitz more or less immediately.

rootzoll, 2020-05-02:
There is a feature for bitcoind in the plans called "assumetxo" that could bootstrap with UTXO snapshots and then validate the history lateron:
https://github.com/jamesob/assumeutxo-docs/tree/2019-04-proposal/proposal

Once this is available, it something the RaspiBlitz will offer.

arcbtc, 2020-05-02:
Could LND in the Blitz be neutrino, then use a local copy of the blockchain, once it is synched?

rootzoll, 2020-05-02:
The RaspiBlitz is depend here what solutions are delivered by bitcoin core to solve this. the Coming "assumetxo" was the one solution targeting this use case  - as far as I heard. What other options are out there I dont know and would be something to discuss with the bitcoin core team.

arcbtc, 2020-05-03:
That's why I thought the neutrino angle could work, bypass core

rootzoll, 2020-05-03:
Having two different BTC backends will mess up stuff up on too many ends, will take a lot of time to implement and there is a solution from bitcoin core in the works anyway . If you install a RaspiBlitz, you decided to run a FullNode - not a light client. If Bitcoin Core can speed that up in the future, that would be great - Im waiting for it and "assumetxo" looks like a good future fit.

arcbtc, 2020-05-03:
Fair enough. I'll go bug people about assumetxo!

Cheers

closed 2020-05-03 by arcbtc