raspiblitz/raspiblitz #1155: Use REST for LNbits

arcbtc, 2020-05-02:
Using gRPC for LNbits and having LND compile the gRPC library is a horrid install experience. 

I also have the feeling some of the sluggish UX is due to gRPC complications, as others I know who are using REST, have had much better experiences.

For now, it makes more sense that the LNbits install uses LND's REST interface, like the other services running on the Blitz.When LND has gRPC precompiled, then we could switch it over.

LNbits can remove the unneeded gRPC dependency, which will greatly improve the Blitz install experience.
https://github.com/lnbits/lnbits/pull/26

rootzoll, 2020-05-02:
tested the PR and already merged. looking good - will be part of v1.5

rootzoll, 2020-05-03:
see: https://github.com/rootzoll/raspiblitz/pull/1156#issuecomment-623192271

closed 2020-05-06 by rootzoll