raspiblitz/raspiblitz #1159: TOR problems - now in error loop

hotvacation, 2020-05-07:
Hi, I'm using rPi 3, raspiblitz v1.4:
Since selecting Tor I've been having a number issues. 
Firstly all but one channel wasn't working and I couldn't 'add peers'. Then used the nyx monitor and saw the Warning:
 [WARN] Before Tor can create a control socket in "/run/tor/control", the directory "/run/tor" needs to exist, and to be accessible only by the user and group account that is running Tor. 
 [WARN] /run/tor is not owned by this user (bitcoin, 1002) but by debian-tor (108). Perhaps you are running Tor as the wrong user?
So i made just the first two changes suggested in issue 998 (github.com/rootzoll/raspiblitz/issues/998)

sudo nano /etc/tor/torrc
User debian-tor
sudo chown -R debian-tor:debian-tor /mnt/hdd/tor
then restarted.
Now I seem to be in a loop where LND isn't running/syncing and menu never appears.
here's the debug log:
https://termbin.com/i01i
Any help would be much appreciated!
Thanks.

openoms, 2020-05-08:
If you read lower down in that issue that command did not help with the warns.
Here is what worked for me, but needs further testing and probably some other changes now with lnd 0.10 improving Tor support:
https://github.com/rootzoll/raspiblitz/issues/998#issuecomment-599960455

The warns are not error messages, Tor is working well in v1.4.

Your debug logs looks ok except bitcoind is not synced yet:   
` progress=0.999998`   
Might be on 1.0 already.

If you have problems after any custom changes the easiest solution is to flash the SDcard again and let the recovery take place.

hotvacation, 2020-05-13:
thanks for the reply and all the support you guys offer.
i have upgraded to v1.5 and all seems to be working now.

(my next problem is trying to connect to zap or zeus over Tor with Orbot - not working at the moment)

lucidprojects, 2020-05-14:
noob question, does the bitcoind need to be fully sync'd before running the LND wallet?

rootzoll, 2020-05-14:
@lucidprojects lnd wallet creation can work before bitcoin is synced, but it in general LND will just be operational as a node once its synced up with bitcoind and if bitcoind is still syncing itself that is a rolling process until both are finished.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll