raspiblitz/raspiblitz #1161: status display has no date or time on it

sneak, 2020-05-08:
There's no way to know if the status display is current or not.   It could have frozen; a date and time display (with seconds) should be incorporated into the LCD layout.

frennkie, 2020-05-08:
@rootzoll what do you think about this..? I could add this to the, BlitzTUI title bar (even had this in an earlier version of BlitzTUI). I had second precision.. But I think Minutes is completely fine.

sneak, 2020-05-08:
Seconds means I don’t have to sit there staring at it for a whole minute to know if it froze or is still running.  Please show seconds.

sneak, 2020-05-08:
If you’re feeling fancy, it could show green or red based on the ntp sync status found in `ntptime` output.

rootzoll, 2020-05-14:
Second to see update rate sounds good.

frennkie, 2020-05-23:
@sneak are you using the normal LCD view or the Touch Screen view? In any case this sounds like a great issue for sending in a Pull Request.

rootzoll, 2020-09-21:
Fixed witrh merge https://github.com/rootzoll/raspiblitz/pull/1565 .. will be part of v1.6.1 release

PR #1565 (2020-09-21): add a 'last refresh' line to infoBlitz
Fix #1161

The change is pretty simple. I'm just nor sure about the exact wording and placement. Any better ideas?

Updated Proposal:

[image]

Old Proposal

[image]
commit 31a0e51f (2020-09-21): Add a 'last refresh' line to infoBlitz (#1565)

fixes #1161
commit 9bdd18cc (2020-09-21): Add a 'last refresh' line to infoBlitz (#1565)

fixes #1161
closed 2020-09-21 by rootzoll