raspiblitz/raspiblitz #1162: Joinmarket ease of use hacks and menu

openoms, 2020-05-08:
Some steps we can take in the RaspiBlitz environment to help JoinMarket usage:

- [ ]  make ssh -X work on windows to have the existing Joinmarket-QT GUI running from the blitz (it is demonstrated to work now) and make some helper dialogs to guide setting up on windows /mac /linux: https://github.com/openoms/bitcoin-tutorials/tree/master/joinmarket#remote-connection-with-ssh--x
Working together in the https://t.me/joinmarketorg channel with https://twitter.com/HamishMacEwan we managed to make the ssh -x method work with windows on his node. I am still to try this as mainly I am using Debian Linux (with which it works well already).
[image]
[image]

- [ ] give some menu options to run the Yield Generator and some other things like creating and importing wallets (to extend this https://github.com/openoms/joininbox).
[image]

- [ ] add shorcuts for some commands and include `TorThisTx` for the `joinmarket` user too.

openoms, 2020-05-10:
On Windows:
* download, install and run XMing with the default settings - https://xming.en.softonic.com/
* open Puttty
Fill in 
* `Host Name`: `RASPIBLITZ_LAN_IP`
* `Port`: `22`
Under `Connection`: 
* `Data` -> `Auto-login username`: `joinmarket`
* `SSH` -> Remote command: `python `
Under `SSH`
* `X11` -> `[x] Enable X11 forwarding`

You can save the setting in `Session` -> `Load. save or delete stored session` -> `Save`

`Open`

In the terminal type:
`python joinmarket-qt.py`

And the Joinamrket-QT GUI will appear on the windows desktop running from your RaspiBlitz.

openoms, 2020-05-10:
Will make a good FAQ entry (can edit as needed):
```
#SHORTCUT COMMANDS you can call as user 'admin' from the terminal

* raspiblitz
Calls the the raspiblitz mainmenu with checks

* menu
* blitz
Calls the main menu.

* repair
calls the repair menu

* check
checks the RaspiBlitz config for errors

* debug
Displays the Debug Logs

* restart
Restarts the RaspiBlitz

* hdmi
SWITCHING VIDEO OUTPUT TO --> HDMI

* lcd
SWITCHING VIDEO OUTPUT TO --> LCD

* manage
Starts lndmanage

* torthistx <rawtx>
Broadcasts a raw transaction (copied from Joinmarket or from Electrum Wallet) through Tor to the Blockstreams API and into the network.
```

rootzoll, 2020-05-27:
@openoms we try to get a v1.6RC1 out soon ... anything on the JoinMarket side you like to add to that release.

openoms, 2020-05-27:
@rootzoll nope. There is no new JM release yet, the `TorThisTx` is already added for the `joinmarket` user and there is no further development with the dialog GUI yet.

openoms, 2020-06-12:
I am stuck on one thing with the Joininbox GUI.
Managed to run the Yield generator with systemd: 
the password is stored temporarily:
https://github.com/openoms/joininbox/blob/master/scripts/menu.sh#L91 
then a python script uses it to input at `stdin` to the the yg-privacyenhanced.py script: 
https://github.com/openoms/joininbox/blob/master/scripts/start.script.py

What needs to be done is to expose and show the output of the `privacyenhanced.py` to be able to see if it is starting successfully or not.
Once started the `privacyenhanced.py` generates a .log file in the `~/.joinmarket/logs` directory, but so far I was unable to capture it's terminal output.

Trying to create a logfile here: https://github.com/openoms/joininbox/blob/master/scripts/start.script.py#L15
but it does not happen.
Please let me know if someone can give me some guidance on this maybe on the hackdays?

openoms, 2020-06-13:
Ended up simplifying the scripts and converting  to bash.
Much more menu option are functional now:
[image]

Planning to add a menu option to give guidance and the necessary command to paste to open the Joinmarket-QT GUI as it was demonstrated before on Linux/Windows/Mac and described here: https://github.com/openoms/bitcoin-tutorials/tree/master/joinmarket#use-the-joinmarket-qt-gui-on-the-desktop

Test the menu on a RaspiBlitz with JoinMarket installed with:
* Run the script in the RaspiBlitz terminal (will install JoinMarket if not active already):
```bash
# download
wget https://raw.githubusercontent.com/openoms/joininbox/master/build_menu_on_raspiblitz.sh
# run
bash build_menu_on_raspiblitz.sh
```
https://github.com/openoms/joininbox#installing-the-menu-on-a-raspiblitz-v15

rootzoll, 2020-06-15:
@openoms can you summarize what is still todo to add this to the v1.6 release.

rootzoll, 2020-06-19:
OK I merged #1266 .. so this is now on final testing status for the v1.6RC2 release.

openoms, 2020-07-01:
updating Joinmarket:
```
sed -i s/v0.6.2/v0.6.3.1/g /home/admin/config.scripts/bonus.joinmarket.sh
sudo config.scripts/bonus.joinmarket.sh off
sudo config.scripts/bonus.joinmarket.sh on
```

rootzoll, 2020-07-24:
@openoms do you consider this feature ready for v1.6 release?

rootzoll, 2020-07-24:
OK tested .. joinmarket reinstalls on update/recocer.

rootzoll, 2020-07-25:
@openoms I will close this issue for the v1.6 release. Please reopen if there is anything left to do .. and just in case we can put out a 1.6.1 with smaller improvements if anything else is missing.

closed 2020-07-25 by rootzoll