raspiblitz/raspiblitz #1163: add SSH authorized_keys symlink

frennkie, 2020-05-09:
After every upgrade I have to redeploy my SSH Public Key to `/home/admin/.ssh/authorized_keys`. I think I would make sense to also have this file (or directory) symlinked to the external harddrive, right?

rootzoll, 2020-05-09:
Makes sense - marked it for the v1.6 release.

openoms, 2020-05-11:
Would like to make the nginx certs persist too.
Currently the users need to reset the certificates in Electrum (delete the old ones on desktop and reinstall.the mobile app after updating), hope it can be avoided.

frennkie, 2020-05-23:
I've now done this on my devices:

```
sudo mkdir /mnt/hdd/admin-ssh
sudo chown admin:admin admin-ssh/
ln -s /mnt/hdd/admin-ssh/ ~/.ssh
```

There did not appear to be too much interest in this `feature`. Also this *kind of* creates a *persistent backdoor* (even if it's for myself). So people might be surprised if we silently add this.

Therefore I will close this request.

closed 2020-05-23 by frennkie