raspiblitz/raspiblitz #1165: ThunderHub on the RaspiBlitz

openoms, 2020-05-09:
Another great  LND Lightning Node Manager in your Browser.  
see https://www.thunderhub.io/  
Install instructions on the RaspiBlitz: https://gist.github.com/openoms/8ba963915c786ce01892f2c9fa2707bc

openoms, 2020-05-19:
https://twitter.com/thunderhubio/status/1262822950971023361
https://github.com/apotdevin/thunderhub/releases

v0.6.0 - Add Server Authentication
Changes
ADD: SSO cookie authentication
ADD: Server accounts authentication
ADD: Switch to turn of third party fetching
ADD: Configure ThunderHub from .env file

rootzoll, 2020-05-27:
If somebody on the next HackSprint is creating a install script for this we can put it into the v1.6 - so far I push this a bit back to v1.7

openoms, 2020-05-27:
TunderHub now allows pre-creating password protected accounts on the server side. Running it for a while now. Everything is there for the install script so should be ready soon.

rootzoll, 2020-06-02:
I merged and fixed some smaller things on the service menu.
But I am not able to open the thuderhub page on port 3010.
The systemd service is active, no error in the logs and port 3010 open ... so where to look next?

rootzoll, 2020-06-03:
Perfect - the patch is finxing it and now it runs on my Blitz too :)
So just some final testing needed once v1.6 gets near to final release and were good.

openoms, 2020-06-05:
very cool GUI for the Faraday stats on v0.7.0: https://twitter.com/thunderhubio/status/1268950899935846400?s=20

and our update function works well too :)

rootzoll, 2020-06-05:
The faraday integration is sweet :)

The basic install script is still 0.6.13 ... @openoms will you upadte to the new version?

openoms, 2020-06-06:
Updating to an even newer version v0.7.1.
That's why I felt the need for the upfate script because iterates so quickly.

My bad it is not really faraday integration, but look-alike stats. Anyway the idea is coming from there and very useful in the GUI!

rootzoll, 2020-06-27:
Updated to 0.8.4 see #1295

rootzoll, 2020-07-24:
OK Thunderhub now also makes a clean reinstall on recover/update. Will be part of v1.6 final. Closing issue.

closed 2020-07-24 by rootzoll