raspiblitz/raspiblitz #1166: LND would not run after update from 1.4 to 1.5 because I had already updated LND to 0.10-beta

redmac68, 2020-05-10:
Not a bug, but it would be great for the 1.5 update script to check current LND version and install LND 0.10 or at least provide an option to install LND 0.10 during initial update process.
I updated LND manually when I worked out what was wrong and all good now.

openoms, 2020-05-10:
I don't think it is possible to tell from the database which version was run before and you remove the previous binary with the SDcard.

If used LND 0.10 on v1.4 (or intstalled any 1.5RC-s) will need to update LND again manually on v1.5 from the UPDATE menu to be compatible with the updated database.
Once that is done the new SDcard restores will use lnd 0.10.

Relevant error in the lnd.conf:
`Unable to open channeldb: channel db cannot revert to prior version`

redmac68, 2020-05-10:
Thanks. that works.

closed 2020-05-10 by redmac68