raspiblitz/raspiblitz #1167: Systemd hardening measures

openoms, 2020-05-10:
In the new RaspiBolt tutorial there are very well commented systemd hardening measures.
We could utilize them in our services too.

https://stadicus.github.io/RaspiBolt/raspibolt_30_bitcoin.html#autostart-on-boot
https://stadicus.github.io/RaspiBolt/raspibolt_40_lnd.html#autostart-on-boot

rootzoll, 2021-04-27:
Actions already taken should be noted in `SECURITY.md`

rootzoll, 2021-08-24:
The system services running has root `bootstrap` & `background` the hardenings were removed because they prevented certain system configs/changes during setup (for example creating fresh ssh host keys) throwing setup into a loop. But for services of additional apps it makes sense to keep the hardenings.

Will be part of v1.7.1rc1 for further intense testing.

rootzoll, 2021-10-09:
since rc1 this wuns stable ... closing for final release.

closed 2021-10-09 by rootzoll