raspiblitz/raspiblitz #1168: "update" torrent not connecting

f0xygit, 2020-05-12:
Raspi3b+ 1TB SSD, v1.5

Using the torrent option to download the blockchain. The "blockchain" torrent is finished downloading but the "update" torrent hasn't been able to connect to the tracker and start downloading. Been a few days now. Status is now at "Tracker:  [could not resolve hostname]"

Does it need seeds or is the tracker down?

ptbtc, 2020-05-12:
Had the exact same issue.

f0xygit, 2020-05-12:
qBittorrent on Windows is downloading both torrents just fine.... though not quickly

rootzoll, 2020-05-12:
For Raspi3b torrent should just be a fallback (that will be removed in upcoming versions) ... have you tried to sync the Bitcoin Blockchain with Bitcoin Core on your laptop and then copying it over to the RaspberryPi over LAN with the COPY option?

f0xygit, 2020-05-12:
Ok.

Not yet. At this point I will just let the torrents finish and use them to help get Bitcoin Core going and then transfer over LAN.

Thanks

Bechev, 2020-05-17:
Downloading now and having the same issue, did it end up working?

f0xygit, 2020-05-17:
It eventually did start the update torrent but it was slow so I just ended up using the copy option. Much faster to download the blockchain elsewhere and copy it to the Pi.

rootzoll, 2020-05-21:
Because the torrent tracker of the upadte torrent file had problems I made a try to change those and hope they might work better. You need to hotfix your v1.5 to get the update:
- if you still try to torrent download - press x (delete old)
- Press CTRL+z to excit to terminal
- Run command: `./XXsyncScripts.sh`
- Run command: `raspiblitz`
- Now start torrent option again

If somebody wants to help out better seeding the update-torrent filer for faster downloads, please seed this torrent from your computer/server: `https://github.com/rootzoll/raspiblitz/raw/v1.5/home.admin/assets/raspiblitz-bitcoin2-2020-01-28-update-c.torrent`

kyott81, 2020-05-23:
Thanks for the hotfix rootzoll, but unfortunatelly for me it did not work. 
So I am going to download locally and send it via LAN.

rootzoll, 2020-05-23:
@kyott81 it was worth another shot, but the Quality of Service of a torrent is hard to ensure, thats another reason why we plan to drop the torrent option #1179 for the future and will focus to make the selfvalidation on the laptop and then copy it over LAN option more easy.

Teegardin, 2020-05-23:
someone please seed

kyott81, 2020-05-24:
I tried a couples of time more --> same result.
So, where I can download the blockchain from my laptop so afterwards it can be copied with the Raspberry Pi via LAN?

limontec, 2020-05-26:
The raspiblitz is using only the tracker udp://tracker.coppersurfer.tk:6969/announce ant it is not working, on computer others trackers is working... so can I download the "update" on computer and copy to raspiblitz? How? Just using the copy option or need to do manually? 

Output:
*** 2) Status Torrent 'update':
               0.0 / 88649.7 MB Rate:   0.0 /   0.0 KB Uploaded:     0.0 MB [ 0%
  Tracker[1:1]: Connecting to udp://tracker.coppersurfer.tk:6969/announce
   raspiblitz-bitcoin2-2020-01-28-update-c

f0xygit, 2020-05-26:
I suppose that would work but again.. it is much easier and faster to just sync Bitcoin core on another computer on your network. Then flash a fresh SD card, start raspiblitz setup, and choose copy option.

limontec, 2020-05-26:
Hmm, I checked the /mnt/hdd/torrent folder:
admin@raspberrypi:/mnt/hdd/torrent $ du -sh *
229G    blockchain
112K    raspiblitz-bitcoin2-2020-01-28-update
11G     raspiblitz-bitcoin2-2020-01-28-update-c

20 minutes later I checked again and the output was:
admin@raspberrypi:/mnt/hdd/torrent $ du -sh *
229G    blockchain
112K    raspiblitz-bitcoin2-2020-01-28-update
12G     raspiblitz-bitcoin2-2020-01-28-update-c

So the download is going on... but the script continue showing
*** 2) Status Torrent 'update':
0.0 / 88649.7 MB Rate: 0.0 / 0.0 KB Uploaded: 0.0 MB [ 0%
Tracker[1:1]: Connecting to udp://tracker.coppersurfer.tk:6969/announce
raspiblitz-bitcoin2-2020-01-28-update-c

nyxnor, 2021-04-18:
Pls close this. Torrent dropped.

closed 2021-04-19 by rootzoll