raspiblitz/raspiblitz #1169: Deserialize or I/O error after downloading blocks from Torrent

adolfosrs, 2020-05-12:
Hey,

I have just finished syncing up my node but I'm getting the following error data corruption message every now and then and sometime the node just restart in a previous state (something like 100 blocks in the past).

I'm wondering if thats something that comes from the torrent download or it was my node the fucked up the data.

```
ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=1334, nPos=120886144)
```

rootzoll, 2020-05-14:
This looks like a disk corruption ... see similar error reported on Bitcoind: https://github.com/bitcoin/bitcoin/issues/5668

Please run from the main menu "REPAIR" > "SOFTWARE" or call `debug` on command line and check the value of `powerWARN` if that on is bigger then 0 - it might be a weak power supply. Otherwise you could format your HDD fresh on another computer and give it another try (or replace it).

vmavromatis, 2020-06-01:
Similar problem here https://termbin.com/9ojp my power seems OK... I literally just finished downloading the torrents....does that mean I need to download the torrents again? :/

rootzoll, 2020-06-01:
Please dont use the torrent anymore  - will get removed from nach raspiBlitz version because its too unreliable. If oyu have a RP4 use SYNC .... and if on old RP3 download and selfvalidate the blockchain on your laptop thru Bitcoin Core and then use the COPY option to copy it over LAN to your RaspiBlitz.

vmavromatis, 2020-06-01:
Cheers @rootzoll I update the README, if that's OK, so that other won't run into this until it's deprecated.

rootzoll, 2020-06-01:
@vmavromatis thanks - merged.

nyxnor, 2021-04-18:
Pls close this.

closed 2021-04-19 by rootzoll

lcharles123, 2025-01-09:
seem just data corruption at a given blockchain block. 
Happened to my litecoind node when it was syncing basic filters, that reads old blk files.
`2025-01-09T05:55:17Z ERROR: ReadBlockFromDisk: Deserialize or I/O error - ReadCompactSize(): size too large: iostream error at FlatFilePos(nFile=379, nPos=26597467)
`
I noted that there is many corrpted blk files, so i restored the corrupted file from my local backup using rsync:
`rsync -lt --no-perms --no-owner --no-group   --checksum --progress --recursive LTC_08_2024/blocks/blk00* ~/.litecoin/blocks` 

It just replaced the differences in the files using --checksum flag.

These blocks get corrupted because i used some mount options to speedup HDD, like nobarrier, and the system crashed with that (maybe because the disk are not healthy). For use the system again it was needed to run fsck, that fixed some inconsistencies, and then creating others, corrupting some blk files.