raspiblitz/raspiblitz #1170: StaticChannelBackup on DropBox not updating

senf42, 2020-05-13:
I activated this service from the service menu, provided the DropBox Authtoken, and the channel.backup showed up as expected. The next day I opened a new channel. It's no longer pending open and I've has some payments route over it, but the channel.backup on DropBox still has the modified timestamp of the upload when I first started the service instead of the time I opened the new channel. 

What can I do to troubleshoot the file not updating? Thanks!

rootzoll, 2020-05-14:
@senf42 you should check the logs of the background script: `sudo journalctl -f -u background` to see why its not updating (you would need to monitor around the time when a channel closes or updates)

I just tested on my node again and it updates the channel.backup on change ... you should see somting like this in the logs:
```
May 14 22:42:19 **** sudo[1147]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/curl -s -X POST https://content.dropboxapi.com/2/files/upload --header Authorization: Bearer KbYkILk****Z5rWLSUlt3qkzz --header Dropbox-API-Arg: {"path": "/lndbackup-*****/channel.backup","mode": "overwrite","autorename": true,"mute": false,"strict_conflict": false} --header Content-Type: application/octet-stream --data-binary @/home/admin/.lnd/data/chain/bitcoin/mainnet/channel.backup
```

But to keep in mind, it might take a while until the opening/closing is not pending anymore before the file gets changed - what is then detected by the background script, which then should trigger the update to dropbox.

senf42, 2020-05-15:
I have this that shows up every now and then when watching the background script:

May 15 01:57:49 **** _background.sh[3035]: --> Offsite-Backup Dropbox
May 15 01:57:49 **** sudo[31153]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/home/admin/config.scripts/dropbox.upload.sh upload 3x3FflK-kMAA****OIUDRRL /home/admin/.lnd/data/chain/bitcoin/mainnet/channel.backup
May 15 01:57:49 **** sudo[31153]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 15 01:57:49 **** sudo[31164]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/curl -s -X POST https://content.dropboxapi.com/2/files/upload --header Authorization: Bearer 3x3FflK-kMAA****OIUDRRL --header Dropbox-API-Arg: {"path": "/lndbackup-****/channel.backup","mode": "overwrite","autorename": true,"mute": false,"strict_conflict": false} --header Content-Type: application/octet-stream --data-binary @/home/admin/.lnd/data/chain/bitcoin/mainnet/channel.backup
May 15 01:57:49 **** sudo[31164]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 15 01:57:50 **** sudo[31164]: pam_unix(sudo:session): session closed for user root
May 15 01:57:50 **** sudo[31243]:     root : TTY=unknown ; PWD=/ ; USER=root ; COMMAND=/usr/bin/shred -u /home/admin/.dropbox.tmp
May 15 01:57:50 **** sudo[31243]: pam_unix(sudo:session): session opened for user root by (uid=0)
May 15 01:57:50 **** sudo[31243]: pam_unix(sudo:session): session closed for user root
May 15 01:57:50 **** sudo[31153]: pam_unix(sudo:session): session closed for user root
May 15 01:57:50 **** _background.sh[3035]: FAIL -  size zero
May 15 01:57:50 **** _background.sh[3035]: name channelbackup pathlower lndbackup****channelbackup pathdisplay lndbackup****channelbackup id idLc****AAFg clientmodified 20200509T161651Z servermodified 20200509T161651Z rev 015a53****814b0 size 0 isdownloadable true contenthash e3b0c44298****852b855

rootzoll, 2020-05-15:
So the upload dropbox gets triggered, but not sure why it does not update it in your dropbox then. Can you copy the command from the log and try to run it with sudo on your terminal and check the result?

senf42, 2020-05-15:
Is that this command? 

sudo /home/admin/config.scripts/dropbox.upload.sh upload 3x3FflK-kMAA(stuff)OIUUDRRL /home/admin/.lnd/data/chain/bitcoin/mainnet/channel.backup

If so, I get this after trying it:

# Upload happened but is size zero
upload=0
err='size zero'
errMore='name channelbackup pathlower lndbackup(stuff)channelbackup pathdisplay lndbackup(stuff)channelbackup id idL(private?)AFg clientmodified 20200509T161651Z servermodified 20200509T161651Z rev 015a(private?)814b0 size 0 isdownloadable true contenthash e3b0c4(private?)52b855'

openoms, 2020-05-16:
Is the size of the channel.backup file zero?
Try  to delete it on Dropbox and/or the RPi.

senf42, 2020-05-16:
Thanks for the help, sorry I'm not super familiar with command line stuff. I put these commands in, seems like there are three files in that directory but no channel.backup? 

admin@(mynode):~ $ cd /home/admin/.lnd/data/chain/bitcoin/mainnet/
admin@(mynode):~/.lnd/data/chain/bitcoin/mainnet $ ls
admin.macaroon  invoice.macaroon  readonly.macaroon

openoms, 2020-05-16:
Did you try restarting the node yet?
You can find the file on all places with:
```
cd /
sudo find -name channel.backup
```

senf42, 2020-05-17:
After restarting, still no channel.backup in /home/admin/.lnd/data/chain/bitcoin/mainnet/ when I try looking there directly. 

That command found these three:
admin@(mynode):/ $ sudo find -name channel.backup
find: ‘./proc/1487’: No such file or directory
./mnt/hdd/lnd/data/chain/bitcoin/mainnet/channel.backup
./mnt/hdd/backup_lnd/data/chain/bitcoin/mainnet/channel.backup
./home/admin/backups/scb/channel.backup

rootzoll, 2020-05-21:
@senf42 I think I found the problem and commited a fix to the v1.5 hotfix branch. Please run the HotPatch: MAINMENU > UPDATE > PATCH > PATCHMENU > PATCH and reboot

Then try to open/close some channels again and let me know if this solved the problem.
Much thanks already for reporting and debugging this with us.

senf42, 2020-05-22:
I patched and rebooted, and put in a channel close at ~20sat/byte. I'll update here when that goes through. 

I'm glad that I can help with reporting and debugging! You're amazing, and I appreciate all you do.

senf42, 2020-05-22:
The closing transaction got picked up, it's at 6 confirmations, and I see an update in DropBox about an hour ago for the channel.backup. I'd say everything looks good! Thanks again!

closed 2020-05-22 by rootzoll