raspiblitz/raspiblitz #1171: Installation of specter-desktop is failing since 11th may

k9ert, 2020-05-13:
Specter-desktop created the 0.3.0 release on 11th may. It updates the decpendency of HWI to 1.1.0.
HWI 1.1.0 is dependendent on pyside2>=5.14.0,<6.0.0
https://github.com/bitcoin-core/HWI/blob/master/setup.py#L24

However, pyside2 is not available on the arm-platform. This results in a failure of the installation of specter-desktop. I hope i can find a solution by making that dependency optional.

In the meantime i'll post a workaround in the next comment soon.

k9ert, 2020-05-13:
After you have installed Cryptoadvance-specter unsuccessfull, leave the raspiblitz-menu or press CTRL-C. On the commandline do this:
```
sudo su - bitcoin
cd .specter
. ./.env/bin/activate
# Install all the dependies of hwi one by one
pip3 install 'ecdsa>=0.13.0,<0.14.0' 'hidapi>=0.7.99,<0.8.0' 'libusb1>=1.7,<2.0' 'mnemonic>=0.18.0,<0.19.0' 'pyaes>=1.6,<2.0' 'typing-extensions>=3.7,<4.0'
pip3 install hwi --no-dependencies
curl https://raw.githubusercontent.com/cryptoadvance/specter-desktop/master/requirements.txt | xargs -n 1 pip install
pip3 install cryptoadvance.specter --no-dependencies
exit
sudo service cryptoadvance-specter restart
```

After that, specter should be available as shown on the menu.

pretyflaco, 2020-05-13:
I can confirm that this solved it and I can run Specter Desktop as described in Menu

rootzoll, 2020-05-14:
@k9ert can you make a PR that will fix the Specter Install-Script `/home.admin/config.scripts/bonus.cryptoadvance-specter.sh`? Thx.

k9ert, 2020-05-18:
I have to fix the issue entirely. What's imho missing in the install-script is that it's not failing but pretending to have suceeded. That's far from gracufully failing.
I could potentially include this workaround in the install-script but that's definitely a workaround, not a solution.

rootzoll, 2020-05-21:
Yes thats the evolution of this install scripts - first make them work, then make them robust. Looking forward to your update of the script ... lets us know if we can help out.

rootzoll, 2020-06-02:
updating to Specter 0.4.0
https://twitter.com/k9ert/status/1267880096003653632?s=20

rootzoll, 2020-06-02:
OK Specter 0.4.0 will now run again with v1.6 release .. please test once we have a RC1 out.
I have fixed the version on install so then pips get new version that it stays working ;)

k9ert, 2020-06-05:
I'm quite sure that 0.4.0 is not yet fixing that issue. Effectively specter-desktop can't fix this issue. It need to be fixed in [HWI](https://github.com/bitcoin-core/HWI/issues/331). This issue is fixed but we need a new HWI-Release to really fix it for us. As soon as that happened, we'll create a new specter-desktop release to finally fix it.

k9ert, 2020-06-05:
I wasn't aware that HWI created a new release. We'll create a new release of specter-desktop this afternoon. Should be fixed soon.

k9ert, 2020-06-06:
We did the release 0.4.1 and i've tested it with the improved installation-script. IMHO we can close this issue. PR is to master. Not sure that was right so.

rootzoll, 2020-06-06:
@k9ert thanks, installation looking good on my machine - will be part of v1.6 release

openoms, 2020-06-19:
Will need two corrrections:
- [ ] the .onion address uses https, but forwarded to to the port 80. Needs to use the port 443 (need to accept the self signed cert even in the Tor browser) 
- [ ] the .onion address does not fit in the status menu. Need to increase the window size

rootzoll, 2020-07-16:
https://github.com/cryptoadvance/specter-desktop/releases/tag/v0.5.5 is out

rootzoll, 2020-07-16:
The pip was already updated: https://pypi.org/project/cryptoadvance.specter and the install script installs the latest ... so give it a test run on RC.

rootzoll, 2020-07-16:
Also check if its working with the nginx https ... because in the install script it still creates its own https

rootzoll, 2020-07-16:
@frennkie specter is still using its own HTTPS cert ... would it make sense to also put it behind the nginx HTTPS proxy instead?

rootzoll, 2020-07-24:
OK tested and reinstalls during update. Will be part of v1.6. Closing issue.

closed 2020-07-24 by rootzoll

ben-kaufman, 2020-08-05:
Hi, if that's not too late to make it for the v1.6 release, we just released [Specter v0.6.0](https://github.com/cryptoadvance/specter-desktop/releases/tag/v0.6.0), which among other things also includes a fix to a potential installation issue with Specter v0.5.5. If that's not too late already it will be really great to have that included in the coming release.
Another new notable thing in this release is that it also includes binaries you could use to run Specter instead of downloading from pip if you find that preferable.

rootzoll, 2020-08-05:
@ben-kaufman there will be a service realease v1.6.1 not long after v1.6 there we can include the newest specter.