raspiblitz/raspiblitz #1172: Node up and running, 4 channels funded with 5m sats, cannot send or receive lightning payments.

sal-btc21, 2020-05-13:
Using RTL when I try and send sats I get this response:

Send Payment Failed
Error Code
500
Error Message
checksum failed. Expected l6jzgr, got fclpnf.

When I try and send sats to my node's wallet from bluewallet I get this:

API error:Payment failed. Does the receiver have enough inbound capacity? (code 10)

I have 4 channels active with a total of 5 million sats.

This has been an issue for 2 weeks since I got the node up and running (tor).
Can anyone help?  I'm a coding novice and don't understand what might be going on here.
Thanks in advance.

frennkie, 2020-05-13:
You say you 4 funded channels... But do you also have inbound capacity? So SAT on the remote side of any of the channels..? If all funds in the channels are on your side you can not receive.

If you spent a certain amount (more a certain base amount) then you will be able to receive from you BlueWallet app.

sal-btc21, 2020-05-14:
Remote total is circa 3k sats.  Anyway, I can't send sats from the node, when I try to I get the checksum error.

frennkie, 2020-05-14:
Well.. 3k is far from 1% of 5m SAT. But you are right that a checksum error indicates another issue.

First I would recommend that you upgrade to the current Raspiblitz Version 1.5.

Then try again. Did you only try using RTL? You should also try either the command line interface or if you are not comfortable with that the RaspiBlitz Menü for sending. This will give an indication whether the problems is caused by RTL or something else.

sal-btc21, 2020-05-14:
I have updated.  Tried using Raspiblitz menu and interestingly, it begins by reporting a FAIL and then the payment goes through....

RESULT (may wait in case of timeout):
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
FAIL
try with a wallet app or the RTL WebGUI (see services)
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
{
    "payment_error": "",
    "payment_preimage": "4aeaa152ae1102954e6975333a6ff7cccb4523672c6c75719cc774fd165a34bf",
    "payment_route": {
        "total_time_lock": 630381,
        "total_fees": "1",
        "total_amt": "101",
        "hops": [
            {
                "chan_id": "691497156516052993",
                "chan_capacity": "1340000",
                "amt_to_forward": "100",
                "fee": "1",
                "expiry": 630341,
                "amt_to_forward_msat": "100000",
                "fee_msat": "1000",
                "pub_key": "02004c625d622245606a1ea2c1c69cfb4516b703b47945a3647713c05fe4aaeb1c",
                "tlv_payload": true,
                "mpp_record": null,
                "custom_records": {
                }
            },
            {
                "chan_id": "672003914739482624",
                "chan_capacity": "124912",
                "amt_to_forward": "100",
                "fee": "0",
                "expiry": 630341,
                "amt_to_forward_msat": "100000",
                "fee_msat": "0",
                "pub_key": "02594c59aceeee247184ffc8abac8213dd773c789e18af2c20c4d1613db82afb0a",
                "tlv_payload": true,
                "mpp_record": {
                    "payment_addr": "08909e0a862caf1d43ff995196382d39537e986b788501d81c1200b325656879",
                    "total_amt_msat": "100000"
                },
                "custom_records": {
                }
            }
        ],
        "total_fees_msat": "1000",
        "total_amt_msat": "101000"
    },
    "payment_hash": "5054dcea479efa7546b6c70a287b384721f2cee47d95740f8a82fbb057406579"
}

sal-btc21, 2020-05-14:
Still can't receive sats though

frennkie, 2020-05-14:
Can you run `lnchannels` and paste the result here..?

sal-btc21, 2020-05-14:
mainnet                            Commit ------- Balance --------- --- Fee ----
Alias or Pubkey           Capacity   Fee        Local       Remote   Base PerMil
-------------------- ------------- ------ ------------ ------------ ----- ------
WalletOfSatoshi.com        1340000  21404      1318596          262  1000      1
nc: getaddrinfo for host "djbeflhgdy7at33da2bgtfz4mdwdqakrkad72e4risi3ildon3rbubid.onion" port 9735: Name or service not known
RaspiBlitzDevHC1           2000000  18264      1981736            0  1000      1
 > No response from Addr:Port djbeflhgdy7at33da2bgtfz4mdwdqakrkad72e4risi3ildon3rbubid.onion:9735
rompert.com🔵              1300000  21148      1277257         1595  1000      1
mynodebtc.com [myNod        200000  21411       178589          268  1000      1
SELFBANKT                   600000  21142       578858            0  1000      1
-------------------- ------------- ------ ------------ ------------ ----- ------
Totals          5 ch       5440000 103369      5335036         2125 Day:       0
                                                                    Week:      0
                                                                    Month:     0

frennkie, 2020-05-14:
Please use the Github markdown code syntax (3x ` ) so this is readable the next time (I made the change).
None of your channels has enough inbound capacity to allow a payment to be received.

There needs to be a channel reserve left after the transaction on the remote side of at least 1% of the channel capacity.

WalletOfSatoshi.com        1340000  262  <- 0.195%
RaspiBlitzDevHC1           2000000  0  <- nothing on remote
rompert.com🔵              1300000  1595  <- 0.122%
mynodebtc.com [myNod        200000  268 <- 0.134%
SELFBANKT                   600000  0 <- nothing on remote

Transfer Move e.g. 20k SAT out of the WalletOfSathoshi channel.. then you would have inbound capacity there..

closed 2020-05-14 by frennkie

sal-btc21, 2020-05-15:
Thanks for your answer