raspiblitz/raspiblitz #1173: Loop install error when updating to lnd 0.10 after recovering from 1.5RC2 with lnd 0.10

openoms, 2020-05-13:
Documenting this error (doesn't seem to break anything):
[image]

Updated a 1.5RC2 with lnd0.10 with a new 1.5 SDcard. 

LND 0.10 needs to be installed manually and the Loop install shows an error ,despite being installed already.

Seems to work for now.

frennkie, 2020-05-14:
Saw it two but forgot to open an issue.

openoms, 2020-05-14:
Loop remained on v0.5.1 (for lnd 0.9.2) where it restored initially and did not update with lnd 0.10.

Reinstalling Loop from the menu solved it and it is now functional on v0.6.0.

rootzoll, 2020-05-14:
Normally when updating LND to 0.10.0 if LOOP was activated tit should get deinstalled and then reinstalled ... but I found a typo. Fixed this.

closed 2020-06-03 by openoms