raspiblitz/raspiblitz #1176: Update ElectRS to 0.8.3

rootzoll, 2020-05-14:
https://github.com/rootzoll/raspiblitz/pull/1175

rootzoll, 2020-06-12:
Also ElectRS is now served over HTTPS with nginx.
- [ ] retest ElectRS over HTTPS

openoms, 2020-07-02:
Electrs 0.8.5 is out (improves Electrum 4.0 compatibilty)

https://github.com/romanz/electrs/blob/master/RELEASE-NOTES.md

rootzoll, 2020-07-03:
OK ElectRS is update
- [ ] retest ElectRS over HTTPS

ogas2, 2020-07-21:
Hi, how can I update my raspiblitz to have the ElectRS 0.8.5?

rootzoll, 2020-07-21:
@ogas2 ElectRS 0.8.5 will be part of RaspiBlitz v1.6 release. There is already a RC3 available for v1.6 ... but if its just recommended for test nodes: https://github.com/rootzoll/raspiblitz/blob/v1.6/README.md#downloading-the-software

Final v1.6 will be out within next week hopefully.

rootzoll, 2020-07-25:
Not sure how to test the https (electrum just eat sthe https string) .. but it installs after recovery/update. Will be part of v1.6. Closing issue.

closed 2020-07-25 by rootzoll