raspiblitz/raspiblitz #1177: Update Loop to v0.6.2-beta

rootzoll, 2020-05-14:
Introducing Multi-Loop
https://twitter.com/lightning/status/1261019739935191047?s=20

rootzoll, 2020-06-03:
TODO: Test if working - also with RTL

openoms, 2020-06-05:
Loop updated to 0.6.3 allowing Loop out from multiple channels with LND 0.10.1. https://github.com/lightninglabs/loop/releases
Surely command line only, but should not break RTL compatibility.

rootzoll, 2020-06-05:
Nice - will add fpr v1.6. With all those releases coming out rapidly .. its like living on a train station :)

openoms, 2020-06-13:
https://github.com/lightninglabs/loop/releases/tag/v0.6.4-beta

new update with some bug fixes

openoms, 2020-06-14:
have the certificate bug in the go install script as well.

```
*** Installing Go v1.13.3 for armv6l ***
--2020-06-14 13:35:03--  https://dl.google.com/go/go1.13.3.linux-armv6l.tar.gz
Resolving dl.google.com (dl.google.com)... ::
Connecting to dl.google.com (dl.google.com)|::|:443... connected.
ERROR: The certificate of ‘dl.google.com’ is not trusted.
ERROR: The certificate of ‘dl.google.com’ doesn't have a known issuer.
The certificate's owner does not match hostname ‘dl.google.com’
!!! FAIL !!! Download not success.
```

easy fix, but TODO

rootzoll, 2020-06-18:
OK updated to 0.6.4 ... installs and looks good so far. Ready for final testing.

rootzoll, 2020-07-24:
Testing worked - together with RTL. Closing issue.

closed 2020-07-24 by rootzoll