raspiblitz/raspiblitz #1178: Make AssumeUTXO new Default (when available)

rootzoll, 2020-05-15:
As soon as AussumeUTXO is availlable this should then become the new default to get started with the blockchain (that even should work with old RP3): https://github.com/jamesob/assumeutxo-docs/tree/2019-04-proposal/proposal

rootzoll, 2020-05-21:
Note about where to get the 3GB UTXO snapshot from to bootstrap with assumeutxo ... simply make it part if the RaspiBlitz sd-card release :)

rootzoll, 2023-11-14:
AssumeUTXO is now ready: https://github.com/bitcoin/bitcoin/pull/27596

fjahr, 2024-08-15:
> AssumeUTXO is now ready

Now it's ready ready, it will be in v28 :)

rootzoll, 2024-09-29:
It would make sense to combine this with introduing BootFromSSD with v1.12.0 .. because if users want to reset their existing SSD from just datastorage to boot & datastorage they might need to migrate data and resync Blockchain - AssumeUTXO could minimize the Blockchain resynctime in this process.

rootzoll, 2024-10-04:
OK created [branch `1178-assumeutxo`](https://github.com/raspiblitz/raspiblitz/tree/1178-assumeutxo) that used bitcoin v28.0 to support default parameters for [assumeUTXO](https://bitcoinops.org/en/topics/assumeutxo/) have been added, allowing the loadtxoutset RPC to be used with a UTXO set downloaded outside of the Bitcoin network (e.g. via a torrent).

Bitcoin v28.0 also supports testnet4 - so that branch is also the base for [#4580](https://github.com/raspiblitz/raspiblitz/issues/4580#issuecomment-2385383192).

geco91, 2024-10-10:
Torrent file here: https://github.com/bitcoin/bitcoin/pull/28553 10GB from last halvening.
Core has to complete header sync before loading UTXO set.
Tow options to detect block header ready state of core:
- Wait until blocksync is not zero anymore
- Wait until `loadtxoutset` does not return with error (quick return with error when not ready)

Error return when not ready:
```
msg = "Unable to load UTXO snapshot: The base block header (3bb7ce5eba0be48939b7a521ac1ba9316afee2c7bada3a0cca24188e6d7d96c0) must appear in the headers chain. Make sure all headers are syncing, and call loadtxoutset again."
```

On `loadtxoutset` call set rpctimeout to 0 (means off) because to load the set can take a long time. Progress might be available thru log parsing.

When ready/done then `getchainstates` delivers to states (snapshot & ibd). If it is ready the snapshot info might stay until bitcoin core is restarted.

Maybe give user the option to seed the snapshot afterwards to support other snapshot users.

geco91, 2024-10-10:
Consider to fatpack the the snapshot - might change recommendation for sd card to 64GB?

rootzoll, 2024-11-24:
Some article for further background:
https://blog.lopp.net/bitcoin-node-sync-with-utxo-snapshots/