raspiblitz/raspiblitz #1180: Fresh Raspiblitz on RPi4 don't boot after creating Swap File and Reboot

Pinkman420, 2020-05-15:
Hi. I have been trying to get Raspiblitz running for 2 days now. But every time swap was created and then the restart was boot no longer. The hard drive is constantly charging and the RaspberryPi does not boot. SSH access is also no longer possible. Have now tried everything, formatted everything several times, another SD card no chance, I can't get Raspiblitz to run. After a new installation of the same game, Raspiblitz boots, you enter all the data, then swap is created, restart bootloop.

Pinkman420, 2020-05-15:
Okay I think i found the issue. Apparently the hard drive cable has a break.

Now the sync is running! :-)

closed 2020-05-16 by rootzoll

Pinkman420, 2020-05-17:
Unfortunately I have to open the problem again.

The problem is still there. After the sync, the RPi restarts and then does not start. The HDD is constantly charging. As soon as I remove the entry for the HDD from the fstab, the RPi boots again. In the syslog nothing stands out because of mount.
I also have no desire to flash the Raspiblitz for the umpteenth time.

An idea?

Pinkman420, 2020-05-17:
Yaih I'm finally a step ahead!! I have now formatted the hard drive directly on the RaspberryPi and have chosen Public instead of Tor this time. Now it works! Finally! :D But I couldn't stop it .. I activated Tor, restarted the RaspberryPi and was again with the same problem. Now, as I said, it works without a gate and it looks as if Tor is causing problems when booting from the HDD. Is there a way that Tor stores its files under /var/lib/tor as normal?

I've only had problems since v1.4 and the same HDD was always used for the blockchain.

rootzoll, 2020-05-22:
Thats odd that you get this just when TOR is activated. From version 1.3 to 1.4 we changed from TOR v2 to v3 ... @openoms any idea what can trigger this? Also @Pinkman420 have you still running the ShimFan? If yes can you remove it and try fresh? .. it was the source of a lot of strange boot problems in the past.

Pinkman420, 2020-05-22:
I then got it to work by formatting the HDD directly via the RPi. It has been running without problems since then. Thank you :)