raspiblitz/raspiblitz #1181: update 1.4 to 1.5 without new SD card because pi is more than just a node

phenomeus, 2020-05-15:
hi everybody, is there a way to update 1.4 to 1.5 without writing a fresh SD card with 1.5 because my rpi4 is more than just a RaspiBlitz 

i have also wireguard, Homebridge and pi hole on it and don't want to set up everything from scratch. 

thank you in advance

rootzoll, 2020-05-15:
There is no clean way  - maybe just update LND and Bitcoind manually with scripts by openoms - but basically you stay at v1.4.

See background on this here: https://github.com/rootzoll/raspiblitz/blob/master/FAQ.md#why-do-i-need-to-re-burn-my-sd-card-for-an-update

phenomeus, 2020-05-17:
Thanks for the answer. Unfortunately I read that already and Still hoped to get a hackY script. 

Can I execute the Manual build Script on that 1.4 pi?

Is there an upgrade way from Version 1.5 for the future releases or will it always be like this?

rootzoll, 2020-05-17:
> Is there an upgrade way from Version 1.5 for the future releases or will it always be like this?

This is the upgrade way ... its the idea: if you add personal extra apps to it, that you take care of making them update secure: make yourself a small install script and store important data that should survice an update to mnt/hdd/app-data

> Can I execute the Manual build Script on that 1.4 pi?

I never tried - if you feel lucky, give it a try and report back.

rootzoll, 2020-06-08:
TODO: Make it possible that the user can set a `custom-installs.sh` script in the `app-data` folder that gets called on upadtes with a fresh sd card. This way at least someone can automize to make custom installs in the pi re-done on on a update.

phenomeus, 2020-06-15:
ok, so I did "wget https://raw.githubusercontent.com/rootzoll/raspiblitz/master/build_sdcard.sh && sudo bash build_sdcard.sh" after I stopped bitcoind and lnd and it went through. nice :) 

after the reboot I still see version 1.4 on the blue screen. now I wait until bitcoin is fully started

phenomeus, 2020-06-15:
hmm, tor is somehow "broken": 2020-06-15T21:34:15Z connect() to 127.0.0.1:9050 failed after select(): Connection refused (111)

if I untick TOR in services and do a reboot I have a running bitcoind. if I tick/ activate TOR again, I get the above error

phenomeus, 2020-07-12:
ok i figured it out and updated to 1.5.1, will try this again to version 1.6 and further, will report back. its possible, but with a lot of tinkering. still, thus I learn a lot how raspiblitz is made, very handy to support here in the future

nyxnor, 2021-05-17:
@rootzoll Can this be added to the future release collector? I think it is important to update with flashing the sd card.

rootzoll, 2021-05-17:
@nyxnor There is already a way to add custom installs script to run on a new update - see: https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#why-do-i-need-to-re-burn-my-sd-card-for-an-update is there anyting more you want from this issue?

nyxnor, 2021-05-17:
@rootzoll Can close this.

closed 2021-05-17 by rootzoll