raspiblitz/raspiblitz #1184: Test Raspbian Buster Lite as base image to reduce bloat

openoms, 2020-05-16:
Most sotftware included in the Raspbian Desktop in not used in the RaspiBlitz.

Will attempt to build the SDCard image with the minimal image and identify what is missing if anything.

For the Odroid build I am using a 149 MB minimal Buster image ( https://www.armbian.com/odroid-xu4/) vs Raspbian desktop 1136 MB and even the Lite is 434 MB (https://www.raspberrypi.org/downloads/raspbian/).

frennkie, 2020-05-16:
If I recall correctly then the switch from Lite to Desktop happened when I added BlitzTUI as this uses the XServer, Display / Window manager and such things. 

Getting it to work was a lot simpler by choosing the big image and selectively removing or disabling stuff.

openoms, 2020-05-16:
RaspiBlitz is running with the Desktop version from the beginning.

I think it is worth trying to identify the required extra, seems simpler to me then finding all the unnecessary packages.

frennkie, 2020-05-16:
Okay.. Then I mixed this up.. But I definitely started on creating a PyQT5 from a reduced image and abandoned the approach.

> I think it is worth trying to identify the required extra, seems simpler to me then finding all the unnecessary packages

Agreed.

openoms, 2020-05-19:
Successfully built the SDcard manually on  Raspbian Buster Lite.
It has half the installed packages compared to the desktop version (~500 vs ~1100) ! 
The LCD is working already so will continue with testing the individual services (especially the TUI).

openoms, 2020-05-20:
As expected there are missing dependencies for the TUI during recovery:
```
Turn ON: LCD ROTATE
rm: cannot remove '/etc/X11/xorg.conf.d/40-libinput.conf': No such file or directory
OK - a restart is needed: sudo shutdown -r now
Provisioning Touchscreen - run config script
Turn ON: Touchscreen
making sure system dependencies are installed
debconf: unable to initialize frontend: Dialog
debconf: (Dialog frontend will not work on a dumb terminal, an emacs shell buffer, or without a controlling terminal.)
debconf: falling back to frontend: Readline
debconf: unable to initialize frontend: Readline
debconf: (This frontend requires a controlling tty.)
debconf: falling back to frontend: Teletype
dpkg-preconfigure: unable to re-open stdin: 
installing BlitzTUI (including python dependencies)
/home/admin/config.scripts/blitz.touchscreen.sh: line 39: /home/admin/python3-env-lnd/bin/pip: No such file or directory
sed: can't read /etc/lightdm/lightdm.conf: No such file or directory
grep: /etc/lightdm/lightdm.conf: No such file or directory
tee: /etc/xdg/lxsession/LXDE-pi/autostart: No such file or directory
sed: can't read /etc/xdg/openbox/lxde-pi-rc.xml: No such file or directory
sed: can't read /etc/xdg/openbox/lxde-pi-rc.xml: No such file or directory
sed: can't read /usr/share/themes/PiXflat/openbox-3/themerc: No such file or directory
sed: can't read /usr/share/themes/PiXflat/openbox-3/themerc: No such file or directory
sed: can't read /etc/xdg/openbox/lxde-pi-rc.xml: No such file or directory
make sure pi is member of lndreadonly and lndinvoice
make sure symlink to central app-data directory exists
LCD is rotated into default - no touchscreen rotate
OK - a restart is needed: sudo shutdown -r now
Provisioning UPS - not active
Provisioning LNbits - run config script
making sure services are not running
*** INSTALL LNbits ***
```

rootzoll, 2020-05-20:
@openoms what dependencies are still missing?

openoms, 2020-05-20:
seems to be PIP issues
added:
```
# make sure /usr/bin/pip exists (and calls pip3 in Debian Buster)
sudo update-alternatives --install /usr/bin/pip pip /usr/bin/pip3 1
```
PR incoming, but building again first on Raspbian Lite

openoms, 2020-05-20:
now installs without errors, but I guess still would need things like
LXDE and lightDM at least

```
Turn ON: Touchscreen
making sure system dependencies are installed
installing BlitzTUI (including python dependencies)
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing ./raspiblitz/home.admin/BlitzTUI
Requirement already satisfied: googleapis-common-protos in ./python3-env-lnd/lib/python3.7/site-packages (from BlitzTUI==0.46.1) (1.51.0)
Requirement already satisfied: grpcio in ./python3-env-lnd/lib/python3.7/site-packages (from BlitzTUI==0.46.1) (1.29.0)
Requirement already satisfied: inotify in ./python3-env-lnd/lib/python3.7/site-packages (from BlitzTUI==0.46.1) (0.2.10)
Requirement already satisfied: psutil in ./python3-env-lnd/lib/python3.7/site-packages (from BlitzTUI==0.46.1) (5.7.0)
Requirement already satisfied: pyqtspinner in ./python3-env-lnd/lib/python3.7/site-packages (from BlitzTUI==0.46.1) (0.1.1)
Requirement already satisfied: qrcode in ./python3-env-lnd/lib/python3.7/site-packages (from BlitzTUI==0.46.1) (6.1)
Requirement already satisfied: protobuf>=3.6.0 in ./python3-env-lnd/lib/python3.7/site-packages (from googleapis-common-protos->BlitzTUI==0.46.1) (3.11.3)
Requirement already satisfied: six>=1.5.2 in /usr/lib/python3/dist-packages (from grpcio->BlitzTUI==0.46.1) (1.12.0)
Requirement already satisfied: nose in ./python3-env-lnd/lib/python3.7/site-packages (from inotify->BlitzTUI==0.46.1) (1.3.7)
Requirement already satisfied: setuptools in ./python3-env-lnd/lib/python3.7/site-packages (from protobuf>=3.6.0->googleapis-common-protos->BlitzTUI==0.46.1) (40.8.0)
Building wheels for collected packages: BlitzTUI
  Running setup.py bdist_wheel for BlitzTUI ... done
  Stored in directory: /tmp/pip-ephem-wheel-cache-6w74ou9h/wheels/f9/0b/dd/649ceef3de574283bea3013c5137570269cca956963db31bd6
Successfully built BlitzTUI
Installing collected packages: BlitzTUI
  Found existing installation: BlitzTUI 0.46.1
    Uninstalling BlitzTUI-0.46.1:
      Successfully uninstalled BlitzTUI-0.46.1
Successfully installed BlitzTUI-0.46.1
```

openoms, 2020-05-20:
Adding lightDM was necessary, more packages:
```
$ sudo apt install lightdm
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  busybox desktop-base fonts-quicksand gnome-accessibility-themes gnome-themes-extra gnome-themes-extra-data gtk2-engines-pixbuf initramfs-tools initramfs-tools-core klibc-utils libindicator3-7 libklibc
  liblightdm-gobject-1-0 libplymouth4 libxklavier16 lightdm-gtk-greeter linux-base pigz plymouth plymouth-label xserver-xorg xserver-xorg-legacy xserver-xorg-video-all xserver-xorg-video-amdgpu
  xserver-xorg-video-ati xserver-xorg-video-fbdev xserver-xorg-video-nouveau xserver-xorg-video-radeon xserver-xorg-video-vesa
Suggested packages:
  gnome | kde-standard | xfce4 | wmaker accountsservice upower xserver-xephyr plymouth-themes firmware-amd-graphics xserver-xorg-video-r128 xserver-xorg-video-mach64
The following NEW packages will be installed:
  busybox desktop-base fonts-quicksand gnome-accessibility-themes gnome-themes-extra gnome-themes-extra-data gtk2-engines-pixbuf initramfs-tools initramfs-tools-core klibc-utils libindicator3-7 libklibc
  liblightdm-gobject-1-0 libplymouth4 libxklavier16 lightdm lightdm-gtk-greeter linux-base pigz plymouth plymouth-label xserver-xorg xserver-xorg-legacy xserver-xorg-video-all xserver-xorg-video-amdgpu
  xserver-xorg-video-ati xserver-xorg-video-fbdev xserver-xorg-video-nouveau xserver-xorg-video-radeon xserver-xorg-video-vesa
0 upgraded, 30 newly installed, 0 to remove and 15 not upgraded.
Need to get 12.4 MB of archives.
After this operation, 29.1 MB of additional disk space will be used.
```

openoms, 2020-05-20:
Installing LXDE seems to defeat the purpose of starting with the Lite image as it adds 464 packages including the desktop tools like libreoffice and media players https://pastebin.com/raw/sh5qtLGk

The experiment is still useful since Armbian has no desktop version and just got my Odroid C4 with a display so curious if I can make the TUI work on that.

frennkie, 2020-05-20:
> Installing LXDE seems to defeat the purpose of starting with the Lite image as it adds 464 packages

/o\

BlitzTUI does not exactly require that LXDE is used... We could investigate other Window, Display and Logon Managers... Actually... No desktop environment is strictly required... If I recall correctly the only hard requirement is the XServer...

openoms, 2020-05-21:
@frennkie getting rid of LXDE could be a good research case for the future, but currently the BlitzTUI relies on it for autostart and login at least.

Good news that Armbian has the LXDE desktop as a built-in option as well.

rootzoll, 2020-05-27:
@openoms and @frennkie is it realistic now to go with the minimal Raspbian already for v1.6 or are we pushing this a bit back?

openoms, 2020-05-27:
@rootzoll this research has given me valuable insights, but currently I don't think it is any practical to base on the minimal image. 
Would need to reinstall most packages which are present in the desktop image so not achieving much while significantly complicating the SDcard build.

closed 2020-05-27 by openoms

tomfishb, 2020-06-26:
Are these dependencies only required for the screen? Would be nice to have a non-screen image that builds without desktop if so.

FreePietje, 2021-02-17:
> Installing LXDE seems to defeat the purpose of starting with the Lite image as it adds 464 packages including the desktop tools like libreoffice and media players https://pastebin.com/raw/sh5qtLGk

I realize this issue is already closed, but don't know a better place to give some hints.
Below I'll show commands with `aptitude` as I prefer that (to f.e. `apt` itself) as it also has features build-in that others don't have (but may be provided by other (apt) tools).
- `lxde` is a meta-package, which install everything and the kitchen-sink. When trying to minimize the size, that isn't the right place to start
- The number of package that gets installed, does not HAVE to be a good indicator of its size. It may be a LOT of (really) small packages.
When you do `$ aptitude install lxde -s` (-s = simulate; you can do this as a normal user ($)), it not only lists the packages it will install, but also a line like this 
`Need to get 61.7 MB of archives. After unpacking 228 MB will be used.`
(This is on my Debian Sid box, where I may already have various packages that would be installed on a clean system).
That'll give you a proper indication on how much disk space will be used.
- `lxde` is also a (complete) Desktop Environment. Maybe you need for your purpose some MUCH smaller, like `i3`. aptitude then tells me `Need to get 2,085 kB of archives. After unpacking 5,330 kB will be used.`
This is likely the complete other end of the spectrum and maybe/probably is not enough, but you should get the idea.
For the rest of this comment, I'll assume sticking to `lxde`, but it applies to other options as well.
- `apt` installs both **Depends** and **Recommends** (and you should do that too, unless you *really* know what you're doing)
- when doing `aptitude show lxde`, you see both a `Depends:` and a `Recommends:` line. In the case of `lxde` you'll see it actually wants to install a complete showroom full of kitchens :-O
- Each *depends* and *recommends* will also have their own *depends* and *recommends*. Turtles all the way down. So you can see how that can add up quickly.
- LXDE uses the `GTK` toolkit (GNOME too). Earlier in this thread I saw `PyQt` mentioned and `Qt` is another toolkit. When you use both, it REALLY adds up. Try to stick to **either** *GTK* or *Qt*.
- To start off from a much smaller base, see if one/two/couple of `lxde`'s *depends*/*recommends* are sufficient for your needs. You very likely do need `lxde-core`. On a system with as few packages as possible already installed, do `$ aptitude install lxde-core` and see what the effect is. You likely want `lxterminal` too and that gets installed as well (through *recommends*). If you do `aptitude show lightdm`, you'll also see this: `Provides: x-display-manager`. You likely will want a display manager, but it doesn't have to be `lightdm`. When you do `aptitude show x-display-manager` you'll see a list of other display managers that can also *provide* that role/functionality
- While you technically can go crazy in finding the absolute minimum set of packages, you'll likely spend (full) days or weeks doing that. You probably shouldn't do that. Some things you can probably safely mix, when you stick to a `lxde` *theme*, it's much more likely to have been tested together and thus work together well.

HTH