raspiblitz/raspiblitz #1185: Research: Alternative/Additional config format

frennkie, 2020-05-16:
At the moment we only use/support KEY=VALUE settings in our config files.

Especially with special values (True, False) and list this is extremely limited.

A good alternative could be https://en.wikipedia.org/wiki/TOML

rootzoll, 2020-05-21:
A first testcase for the new config value format can be tested with the tor2ip bridge persitence: #1194

Also bash scripts would need to be able to easily read and import the config data.

frennkie, 2020-05-22:
I had a look at these two files that we are writing:

* `/home/admin/raspiblitz.info`
* `/mnt/hdd/raspiblitz.conf`

Unfortunately this format is not directly compatible with `toml`. Here some examples lines:

```
network=bitcoin
chain=test
autoNatDiscovery=off
runBehindTor=on
touchscreen=1
lcdrotate=1
lndExtraParameter='--accept-keysend'
```

If I change it so that the **all** strings are enclosed with single quotes then I can directly parse it with `toml` and it still valid and the same in bash scripts.

```
network='bitcoin'
chain='test'
autoNatDiscovery='off'
runBehindTor='on'
touchscreen=1
lcdrotate=1
lndExtraParameter='--accept-keysend'
```

How to test:

```
apt-get install python3-toml
python -c "import toml; print(toml.load('/mnt/hdd/raspiblitz.conf'))
```` 

Trying to use `source` in  a shell scripts (on a `toml` *native* config file) fails even if only strings and numbers are used a `toml` by convention uses `key = value`. But a simple `sed` enables this (this might fail if there are more complex keys):

```
source <(sed 's/ = /=/' /tmp/foobar.toml)
```

This is a `toml` sample from https://github.com/toml-lang/toml

```
# This is a TOML document.

title = "TOML Example"

[owner]
name = "Tom Preston-Werner"
dob = 1979-05-27T07:32:00-08:00 # First class dates

[database]
server = "192.168.1.1"
ports = [ 8001, 8001, 8002 ]
connection_max = 5000
enabled = true

[servers]

  # Indentation (tabs and/or spaces) is allowed but not required
  [servers.alpha]
  ip = "10.0.0.1"
  dc = "eqdc10"

  [servers.beta]
  ip = "10.0.0.2"
  dc = "eqdc10"

[clients]
data = [ ["gamma", "delta"], [1, 2] ]

# Line breaks are OK when inside arrays
hosts = [
  "alpha",
  "omega"
]
```

rootzoll, 2020-05-27:
I think to have it for `mnt/hdd/app-data/subscriptions.subscriptions.toml` at the moment is fine.

@frennkie this renminds me now that we get more and more py-scripts that also use TOML we need to install this correctly so that also admin & root user can use the scripts without a vertual env ... lets discuss this shortly in video chat if possible

rootzoll, 2020-06-04:
toml lib is now part of raspiblitz global pip - virtual env for standard scripts not needed anymore

closed 2020-06-04 by rootzoll