raspiblitz/raspiblitz #1186: Full node interface for standard banking software via FinTS

INSG2020, 2020-05-16:
**A full node is the equivalent of a private bank.**

It has different bank accounts depending on which wallets or programs are used on the node, e.g:
1 wallet bitcoin core
1 wallet lightning lnd onchain
1 wallet lightning lnd offchain
1 wallet btcpayserver internal
etc.

Especially for merchants easy bookkeeping is essential if they want to use bitcoin payments, but also as a private user it´s easy to get lost with different wallets and balances.

BTCPay server has options to export invoices (manual actions most bookkeepers charge expensive) or to connect to QuickBooks Online, which is quite nice:
https://docs.btcpayserver.org/Accounting/

But there are lots of different home banking and accounting programs (e.g. https://shop.lexware.de/persoenliche-finanzen or https://shop.lexware.de/buchhaltung). In Europe all of these use a standard protocol FinTS to connect to the bank and accounts, manage transfers, get balances etc.:
https://en.wikipedia.org/wiki/FinTS
https://www.hbci-zka.de/spec/4_1.htm (sorry it´s in german)

To give merchants, bookkeepers and normal end users the same user experience like they have with their ordinary bank accounts they should be able to connect to their private bank (full node) via FinTS get a list of accounts and be able to import all movements from these accounts into their finance software (sending transfers would be even better, but we leave that for later).

So what is needed is a server side integration of FinTS (normally done by banks), which connects to the different application like bitcoin core, BTCPay or lnd.

On the client side there are different open source projects of the FinTS protocol available:
https://python-fints.readthedocs.io/en/latest/
https://github.com/mrklintscher/libfintx
https://github.com/nemiah/phpFinTS

But on the server side there doesn´t seem to be an open source solutions, only commercial ones like this: https://subsembly.com/fints-server.html

Something like this could be implemented maybe in a hackathon?
Supported applications on the server side could be added step by step, maybe starting with lnd and btcpayserver?

A FinTS protocol interface for full nodes would be huge step to better integrate Bitcoin/Lightning in standard finance software.

rootzoll, 2020-05-16:
It would really be great to be able to manage your node within your standard bookkeeping software just like a normal foreign currency account by supporting the FinTS. For merchants that want to make Bitcoin on Lightning a normal part of their business - such interfaces are highly needed, so that accounting can be automated as much as possible. 

@INSG2020 I marked this issue for the next lightning hacksprint in june. Will you be available to mentor on this project? And is anybody interested in attaching a bounty to this?

rootzoll, 2020-05-16:
I would recommend to implement such a FinTS interface on a node wrapper API like Lighter https://gitlab.com/inbitcoin/lighter ... this way it can be used with all major Lightning implementations - not just LND and not just for the RaspiBlitz - but happy to have it on the Blitz first :)

INSG2020, 2020-05-18:
Yes, it´s possible to mentor the project on a project management level, less on technical details :-)
Which date is the hacksprint in June?

How does the bounty work? Please send a quick DM on Twitter, also regarding other questions, which are off-topic here.

INSG2020, 2020-05-18:
Compatibility with https://github.com/BlueWallet/LndHub would be also nice.
But maybe in a later version, because this adds the complexity of the private bank having different "customers" with different accounts, not only different accounts.

INSG2020, 2020-06-13:
On the client side open source programs like the following might be used for testing:
https://gnucash.org/index.phtml?lang=en_US
https://www.aquamaniac.de/rdm/
https://kmymoney.org/

There are also 30 days free versions of the programs from Lexware, which are widely used, for example: FinanzManager
https://shop.lexware.de/persoenliche-finanzen

drmartinberger, 2022-10-03:
@rootzoll interested in this bounty!

rootzoll, 2022-10-04:
@drmartinberger I wrote you a DM on telegram. If thats not you, please let me know here.

rootzoll, 2022-10-06:
>Question: What protocol version should be supported by Raspiblitz? 4.1 seems not to be wide-spread.

Basically what gets accepted by a common bookkeeping program - if 3.0 still wortks thats fine.
Lastinfo I got is that https://shop.lexware.de/persoenliche-finanzen has a 30 day free trial  - but havent checked.

rootzoll, 2022-10-06:
@INSG2020 are you still around to help on testing?

rootzoll, 2023-02-26:
- [ ] create install script for v1.9.0 of this FinTS impelementaton: https://github.com/drmartinberger/FueliFinTS (see branch lnbits-via-mt940)

rootzoll, 2023-03-10:
First version of install script done - still TODOS:
- [ ] use master branch when lnbits-via-mt940 is merged
- [ ] db is droped on every install/update - is that a problem?
- [ ] is nginx needed for HTTP-proxy/HTTPS? Removed for now

rootzoll, 2023-03-23:
Thanks for the tutorial: https://medium.com/@martin.berger/a-tutorial-to-connect-gnucash-to-ln-fints-of-your-bitcoin-lightning-node-50ee803e9101

On testing on raspiblitz I could see that the SSL service was running on port 3111 - but but when i tried to connect with GNU cash I got the error:

```
10:30:59 TLS Handshake Error: -54 (Error in the pull function.)
10:30:59 Could not connect to server
```

Checking the `/home/fints/log/fuelifints.log` I could see:

```
2023-03-23 09:29:38,892 INFO  [Thread-1] [] net.petafuel.fuelifints.communication.BlockingSocketListenerThread - SocketServer listening on port 3110
2023-03-23 09:29:40,326 INFO  [Thread-2] [] net.petafuel.fuelifints.communication.BlockingSSLRequestListenerThread - SSLServer listening on port 3111
2023-03-23 09:29:40,328 INFO  [Thread-2] [] net.petafuel.fuelifints.communication.BlockingSSLRequestListenerThread - Accepting new incoming connection
2023-03-23 09:29:58,626 INFO  [Thread-4] [] net.petafuel.fuelifints.communication.BlockingSocketListenerThread - New socket connection thread
2023-03-23 09:29:58,713 ERROR [Thread-4] [] net.petafuel.fuelifints.communication.BlockingSocketListenerThread - Exception parsing responseSize as integer
java.lang.NumberFormatException: For input string: "d`ø¶þ_X¿¸"
	at java.lang.NumberFormatException.forInputString(NumberFormatException.java:65) ~[?:?]
	at java.lang.Integer.parseInt(Integer.java:638) ~[?:?]
	at java.lang.Integer.parseInt(Integer.java:770) ~[?:?]
	at net.petafuel.fuelifints.communication.BlockingSocketListenerThread$SocketRequestHandler.handle(BlockingSocketListenerThread.java:125) [fints-fat.jar:?]
	at net.petafuel.fuelifints.communication.BlockingSocketListenerThread$SocketRequestHandler.access$300(BlockingSocketListenerThread.java:90) [fints-fat.jar:?]
	at net.petafuel.fuelifints.communication.BlockingSocketListenerThread$SocketWorkerThread.run(BlockingSocketListenerThread.java:71) [fints-fat.jar:?]
2023-03-23 09:29:58,747 ERROR [Thread-4] [] net.petafuel.fuelifints.FinTSServer - UncaughtException in Thread[Thread-4,5,main]
java.lang.StringIndexOutOfBoundsException: offset 23, count 3, length 22
	at java.lang.String.checkBoundsOffCount(String.java:3304) ~[?:?]
	at java.lang.String.<init>(String.java:505) ~[?:?]
	at net.petafuel.fuelifints.FinTSVersionSwitch.checkFinTSVersion(FinTSVersionSwitch.java:42) ~[fints-fat.jar:?]
	at net.petafuel.fuelifints.FinTSVersionSwitch.selectController(FinTSVersionSwitch.java:24) ~[fints-fat.jar:?]
	at net.petafuel.fuelifints.communication.BlockingSocketListenerThread$SocketRequestHandler.handle(BlockingSocketListenerThread.java:133) ~[fints-fat.jar:?]
	at net.petafuel.fuelifints.communication.BlockingSocketListenerThread$SocketRequestHandler.access$300(BlockingSocketListenerThread.java:90) ~[fints-fat.jar:?]
	at net.petafuel.fuelifints.communication.BlockingSocketListenerThread$SocketWorkerThread.run(BlockingSocketListenerThread.java:71) ~[fints-fat.jar:?]
```

@drmartinberger any idea what is still missing?

rootzoll, 2023-05-22:
We will ship the FinTS install script with RaspiBlitz v1.9.0 as it is at the moment. It always pulls the latest code on install from the fints github repo - so that can be used as a after-v1.9.0 release patch mechanism.

Just as an TODO I push this to milestone v1.10.0 for now to recheck if there is anything still to fix on the install script and to improve documentation with next release.

closed 2023-09-14 by rootzoll