raspiblitz/raspiblitz #1188: Torrent 'update' wont start

currenc-group, 2020-05-17:
After downloading bitcoin blockchain, the bitcoin update wont start. Any ideas why?
[image]

Pinkman420, 2020-05-17:
same here. Even 3 days later is not possible. Then synced from the PC.

rootzoll, 2020-05-17:
One of the linked tracker services justseed.it went offline forever and also the other 3 backup tracker seemed not reachable for my seeding torrent client.

rootzoll, 2020-05-17:
Planning to drop the torrent download option for the next release - also for other reasons: see #1179 ... please abort the torrent download with x and use the COPY option.

closed 2020-05-17 by rootzoll

currenc-group, 2020-05-18:
It is difficult to find a laptop with more than 250 gb space, so it will be difficult to download it to laptop and copy it to raspberry. Is there any alternative?

Pinkman420, 2020-05-18:
Torrents currently work in uTorrent. I don't know how it is directly via Raspiblitz.

I am currently uploading the blockchain to my server. This may take 2-3 days, but may help users in the future.

rootzoll, 2020-05-21:
Because the torrent tracker of the upadte torrent file had problems I made a try to change those and hope they might work better. You need to hotfix your v1.5 to get the update:
- if you still try to torrent download - press x (delete old)
- Press CTRL+z to excit to terminal
- Run command: `./XXsyncScripts.sh`
- Run command: `raspiblitz`
- Now start torrent option again

If somebody wants to help out better seeding the update-torrent filer for faster downloads, please seed this torrent from your computer/server: `https://github.com/rootzoll/raspiblitz/raw/v1.5/home.admin/assets/raspiblitz-bitcoin2-2020-01-28-update-c.torrent`

Pinkman420, 2020-05-21:
Yup, seeding is going much better now :)

currenc-group, 2020-05-22:
> Because the torrent tracker of the upadte torrent file had problems I made a try to change those and hope they might work better. You need to hotfix your v1.5 to get the update:
> 
> * if you still try to torrent download - press x (delete old)
> * Press CTRL+z (or CTRL+c) to exit to terminal
> * Run command: `./XXsyncScripts.sh`
> * Run command: `raspiblitz`
> * Now start torrent option again
> 
> If somebody wants to help out better seeding the update-torrent filer for faster downloads, please seed this torrent from your computer/server: `https://github.com/rootzoll/raspiblitz/raw/v1.5/home.admin/assets/raspiblitz-bitcoin2-2020-01-28-update-c.torrent`

Thanks mate! I have accepted to wait for multiple months if necessary and went with with sync option. Since torrent is available again, i'd like to cancel the sync and download the blockchain via torrent again. Thing is, i can't cancel the syncing with X anymore.

Is there a way to get back to downloading via torrent, or do i need to re-install everything?

Pinkman420, 2020-05-22:
I was able to choose another option after restart or continue syncing.

currenc-group, 2020-05-22:
> I was able to choose another option after restart or continue syncing.

I didn't get this option unfortunately, it went back to syncing (and lost 0.5% for some reason).
Hopefully rootzoll can give advice.
[image]

rootzoll, 2020-05-22:
Whe it goes backwards it means prob that somehow the validation stopped and because the blockchain growth you get a reverse status %. Not sure why this is happening .. maybe try a reboot: ssh in, CTRL-x or CTRL-z to get to terminal and run command `restart`

currenc-group, 2020-07-13:
After 2 months of syncing, we're almost there 😄 
[image]