raspiblitz/raspiblitz #1190: ELECTRS STATUS says still building index when it's not

mandreko, 2020-05-17:
I just rebuilt my Raspiblitz system since I thought maybe an upgrade from 1.4 to 1.5 had issues. I'm now on a fresh install of 1.5 after doing the RESET ALL option that kept my blockchain.

I enabled the ElectRS service, rebooted, and it started building the index a few days ago.  The size of the files in /mnt/hdd/app-storage/electrs/db stopped growing, and is now at roughly 58GB. I can connect to my node via Electrum by specifying the server manually. But the ELECTRS STATUS screen still says:

```
Electrum server is still building its index.
Please wait and try again later.
This can take multiple hours.
```

As far as I can tell, the indexes are built, and everything is functioning, although I did have to regenerate the nginx ssl cert files with the script in ~/config.scripts. I'm not familiar enough with bitcoind and electrs to troubleshoot it, but it appears that the detection in ~/config.scripts/bonus.electrs.sh that is commented as "experimental" may not be 100%. I'm just not smart enough at this point to identify why.

openoms, 2020-05-20:
Try restarting the node again. It should pick up that Electrs is now synced.

mandreko, 2020-05-21:
This did eventually work. Weird that it didn't pick it up without the restart. Something is still a little quirky. 

Either way, I was able to get the onion address of my electrs service without the ELECTRS->STATUS page, so it didn't cause any problems for me, but thought it was worth mentioning. 

Thanks!

rootzoll, 2020-05-21:
@mandreko thanks for reporting, we will put this on our list to improve for the next release.

openoms, 2020-05-26:
run this:
`sed -i s/v0.8.0/v0.8.3/g /home/admin/config.scripts/bonus.electrs.sh`
before reinstalling electrs from the menu, to get the new version.

mandreko, 2020-05-28:
reinstall of 0.8.3 worked fine with that update. 

It still has the issue that the ELECTRS->STATUS thinks that it's still building it's index when it appears to be fully updated. But I can connect from Electrum with no problem.

openoms, 2020-05-28:
@mandreko can you please share what hw you are running on? RPi3/4? HDD/SSD?

Smiggel, 2020-05-31:
Had an similar issue. It had the index build. Everything was running fine and then suddenly it stopped out of the blue earlier today. I missed a lot of files. Then it build a few new files with files missing and stopped again.

Process was not killed, but it suddenly just stopped indexing.

I run a Rpi4, with a 1TB HDD. BTC, LND en BTC/LND Explorer are running just fine.

mandreko, 2020-06-01:
@openoms 

I have pretty much the suggested build:

Raspberry Pi v4
Sandisk 1TB SSD
480x320 TFT LCD display

rootzoll, 2020-06-03:
During the initial sync the port 50001 is not responding - can be tested with:
```
echo exit | telnet 127.0.0.1 50001 2>/dev/null | grep -c "Connected to"
``` 
returns `0` ... waiting to finish sync on my test device to see if I get a `1` once synced. This test can be combined with the current sync detection to get rid of the false-positives on 'building index' signaling.

Smiggel, 2020-06-03:
In my case Electrum server stops syncing. I am not sure why. It just doesn't continue anymore. Is there anyway I can see som logs from Electrum?

openoms, 2020-06-03:
> In my case Electrum server stops syncing. I am not sure why. It just doesn't continue anymore. Is there anyway I can see som logs from Electrum?

Notes from the manual setup:
https://github.com/openoms/bitcoin-tutorials/tree/master/electrs#monitor-electrs

Do you have enough space on your disk? 
`df -h`
500 GB is not enough for a new sync any more.

Smiggel, 2020-06-03:
> Notes from the manual setup:
> https://github.com/openoms/bitcoin-tutorials/tree/master/electrs#monitor-electrs
>
> Do you have enough space on your disk?
> `df -h`
> 500 GB is not enough for a new sync any [...]

I have a 1TB attached and still have 480GB of free space left. So I think that it should be enough. :-)

Last lines in the electrum log:

`Jun 03 20:32:18 raspberrypi sudo[20058]:     root : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl -u electrs --no-pager -n100
Jun 03 20:32:20 raspberrypi electrs[2682]: 2020-06-03T20:32:20.182+02:00 - TRACE - RPC PeriodicUpdate
Jun 03 20:32:22 raspberrypi electrs[2682]: 2020-06-03T20:32:22.714+02:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":38}\n")
Jun 03 20:32:25 raspberrypi electrs[2682]: 2020-06-03T20:32:25.902+02:00 - TRACE - RPC PeriodicUpdate
Jun 03 20:32:30 raspberrypi sudo[20219]:     root : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/cat /mnt/hdd/tor/electrs/hostname
Jun 03 20:32:31 raspberrypi electrs[2682]: 2020-06-03T20:32:31.655+02:00 - TRACE - RPC PeriodicUpdate
Jun 03 20:32:32 raspberrypi electrs[2682]: 2020-06-03T20:32:32.724+02:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":39}\n")
Jun 03 20:32:34 raspberrypi sudo[20327]:    admin : TTY=pts/0 ; PWD=/mnt/hdd/app-storage/electrs/db/mainnet ; USER=electrs ; COMMAND=/usr/bin/lsof -i
Jun 03 20:32:34 raspberrypi sudo[20327]: pam_unix(sudo:session): session opened for user electrs by admin(uid=0)
Jun 03 20:32:34 raspberrypi sudo[20327]: pam_unix(sudo:session): session closed for user electrs
Jun 03 20:32:37 raspberrypi electrs[2682]: 2020-06-03T20:32:37.390+02:00 - TRACE - RPC PeriodicUpdate
Jun 03 20:32:42 raspberrypi electrs[2682]: 2020-06-03T20:32:42.730+02:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":40}\n")
Jun 03 20:32:43 raspberrypi electrs[2682]: 2020-06-03T20:32:43.047+02:00 - TRACE - RPC PeriodicUpdate
Jun 03 20:32:44 raspberrypi sudo[20844]:       pi : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/home/admin/config.scripts/bonus.electrs.sh status
Jun 03 20:32:44 raspberrypi sudo[20851]:     root : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/systemctl status electrs --no-page
Jun 03 20:32:44 raspberrypi sudo[20862]:     root : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/systemctl status electrs --no-page
Jun 03 20:32:44 raspberrypi sudo[20873]:     root : TTY=tty1 ; PWD=/home/pi ; USER=root ; COMMAND=/bin/journalctl -u electrs --no-pager -n100
Jun 03 20:32:48 raspberrypi electrs[2682]: 2020-06-03T20:32:48.772+02:00 - TRACE - RPC PeriodicUpdate
Jun 03 20:32:52 raspberrypi electrs[2682]: 2020-06-03T20:32:52.737+02:00 - TRACE - RPC Request("{\"jsonrpc\":\"2.0\",\"method\":\"server.ping\",\"params\":[],\"id\":41}\n")
Jun 03 20:32:53 raspberrypi sudo[20979]:    admin : TTY=pts/0 ; PWD=/mnt/hdd/app-storage/electrs/db/mainnet ; USER=root ; COMMAND=/bin/journalctl -n 10000
`

rootzoll, 2020-06-04:
@Emzy do you have a clue whats going on here?

openoms, 2020-06-04:
@Smiggel not sure what the problem is, but you could reinstall with the latest version (see above) and rest the Electrs database in the REPAIR menu to build form scratch.

Smiggel, 2020-06-04:
@rootzoll @openoms I looked at the documentation. I stopped and restarted Electrum. It is running now and says it's done building the index. I will keep an eye out for it. 

Resetting the database using the repair option resulted in the problem above. I tried that before when I stopped building the index.

rootzoll, 2020-06-08:
OK ... for the v1.6 I added an improved detection if electrum is still on initial sync or is just having a short catchup (what was causing the 'still building' state). Test look good so far on my test machine. This is not addressing any problems during the inital sync like some described on thrs thread - this is something for the future to look out for. But for the v1.6 I think the issue can be closed - setting to final testing for now.

Smiggel, 2020-06-09:
Issue seems ok here now. Status screen does say that it's indexing sometimes. I guess that is the same with the topic start. Other then that, it runs fine now.

closed 2020-07-24 by rootzoll

StellarStoic, 2023-01-21:
I'm experiencing the same issue. electrs have nothing more to sync but it is still complaining that is synching. I did reboot the node and also the electrum service but the problem is still there. 

[image]

[image]

[image]

woeisme, 2023-11-26:
I seem to be having the same situation:

Raspiblitz  v0.9.0
bitcoin       v24.0.1
LND           0.16.3-beta
electrs       v0.9.11

the netcat command above returns '1'
but status page just says "Electrum: Building Index (please wait)"

It's been running for 15 hours now
RPI4 with 4Gb 

Any help for which commands to run to get the above output to check status would be appreciated.

rootzoll, 2023-11-27:
> Any help for which commands to run to get the above output to check status would be appreciated.

`/home/admin/config.scripts/bonus.electrs.sh status`