raspiblitz/raspiblitz #1192: Installing BTC Pay Server on Pi 3B

Arno-GitHub, 2020-05-18:
When I upgraded to v1.5 enabeling the BTC Pay Server was not successful.
There were timeout messages when compiling.
My guess is it has to do with CPU power not being enough.

The solution for me was splitting the install script in three parts:
dotnet, nbxplorer, btcpayserver.

Do you have any thoughts on this? Is the Pi 3B still supported?

openoms, 2020-05-19:
You will have memory is issues running it as well. I wouldn't recommend running BTCpayServer with only 1 GB RAM.

The RPi3 is supported as far as the basic Bitcoind and LND setup goes. Works with non resource intensive applications as well like JoinMarket.

closed 2020-05-21 by rootzoll

Pinkman420, 2020-05-22:
Yes I can confirm. On the RPi3, the installation stops at some point. No problems on the RPi4 with 4GB Ram.
A RPi4 with 2GB are also enough.