raspiblitz/raspiblitz #1196: [Hardware Alternative] HardKernel Odroid C4

openoms, 2020-05-21:
I had this promising SBC arrive and will test the RaspiBlitz scripts on it in the coming days: https://twitter.com/openoms/status/1253328059354116096

Advantages over the RPi4 4GB:
- close to fully open source hardware with all schematics readily available: https://wiki.odroid.com/odroid-c4/hardware/hardware#schematics_drawings_and_s905x3_datasheet 
- 64 bit Linux support by default
- eMMC socket - faster and more reliable then SDcards
- faster CPU
- 4*USB3 + microUSB connector
- more robust 12V power supply
- comes with a heatsink by default
- price is comparable, cheaper in many places

Benchmarks:
https://github.com/ThomasKaiser/sbc-bench/blob/master/Results.md:
[image]
[image]
[image]

Many Linux images available and all are 64 bits:
- Debian Buster
- Ubuntu
- Armbian (a choice of Ubuntu or Debian base)
- Manjaro (Arch)

Will start the first tests with Armbian since that is the closest to Raspbian and has Xserver and LXDE desktop option for the Touch UI as discusssed in: https://github.com/rootzoll/raspiblitz/issues/1184

openoms, 2020-05-21:
The LXDE desktop is running on the HDMI already on Armbian Buster.
Screen activation:
https://wiki.odroid.com/accessory/display/3.2inch_tft_touchscreen_shield/c1/start
Touchscreen activation: https://wiki.odroid.com/accessory/display/3.2inch_tft_touchscreen_shield/c1/touch
Autologin: https://wiki.odroid.com/accessory/display/3.2inch_tft_touchscreen_shield/c1/autoconsole

rootzoll, 2020-05-27:
@openoms anything to keep in mind for a v1.6 release on this?

openoms, 2020-05-27:
@rootzoll I have some minor refactoring the sd card build script, but it should not affect buillding on raspbian so there nothing to wait for here.

openoms, 2020-05-28:
The display works so far, waiting to test the TUI.

Install steps:
```
https://wiki.odroid.com/odroid-c4/os_images/ubuntu/mate/20200422
Username 	Password 
root 	odroid
odroid 	odroid 

# activate display https://wiki.odroid.com/accessory/display/3.2inch_tft_touchscreen_shield/c4/start

sudo apt update && sudo apt full-upgrade

# Enable SPI Driver
# https://wiki.odroid.com/odroid-c4/application_note/gpio/spi#tab__odroid-c42
# SPICC0
sudo fdtput -t s /media/boot/meson64_odroidc4.dtb /soc/cbus@ffd00000/spi@13000 status "okay"
 # SPIDEV0
sudo fdtput -t s /media/boot/meson64_odroidc4.dtb /soc/cbus@ffd00000/spi@13000/spidev@0 status "okay"

# disable spidev node to avoid conflict.
sudo fdtput -t s /media/boot/meson64_odroidc4.dtb /soc/cbus@ffd00000/spi@13000/spidev@0 status "disabled"

# Enable LCD and Touchscreen
sudo fdtput -t s /media/boot/meson64_odroidc4.dtb /soc/cbus@ffd00000/spi@13000/ads7846@1 status "okay"
sudo fdtput -t s /media/boot/meson64_odroidc4.dtb /soc/cbus@ffd00000/spi@13000/hktft32@0 status "okay"

reboot
```

then on every start:
```
#Run Console
#map the console number 1 to the frame buffer 4 screen
sudo con2fbmap 1 4
# change the current console to 1
sudo chvt 1
```
https://twitter.com/openoms/status/1265598511523004421

The LCD goes to sleep after a while. Can be woken by pressing a key on an attached keyboard. Did not figure out how to keep awake yet.

openoms, 2020-06-04:
Trying the Odroid forum about controlling the display: https://forum.odroid.com/viewtopic.php?f=205&t=39088

openoms, 2020-06-04:
Building on the eMMC vs using an SDcard is significantly faster. Comparable to using the SSD: https://github.com/rootzoll/raspiblitz/issues/1235

rootzoll, 2020-06-04:
#1243

openoms, 2020-06-05:
The display comes alive automatically with these lines before the SCRIPT activation in `/home/pi/.bashrc`:
```
#map the console number 1 to the frame buffer 4 screen
sudo con2fbmap 1 4
# change the current console to 1
sudo chvt 1
```

As described in: https://forum.odroid.com/viewtopic.php?f=205&t=39088
Editing `/media/boot/boot.ini` with adding 
`consoleblank=0`
within the double quotes after
`setenv bootargs`
keeps the screen alive!

Now need to debug the installation of:
- [ ] touchscreen
- [ ] lndmanage
- [ ] specter
- [ ] lnbits
and 
- [ ] reduce the display font

openoms, 2020-06-23:
Specter needs some debugging
```
Cryptoadvance Specter Setting changed ..
#    --> INSTALL Cryptoadvance Specter ***
#    --> Enable wallets in Bitcoin Core
#    --> Installing prerequisites
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package libusb-1.0.0-dev
```

openoms, 2020-06-24:
Install python 3.7 from source:
```
sudo apt-get install build-essential checkinstall
sudo apt-get install libreadline-gplv2-dev libncursesw5-dev libssl-dev \
    libsqlite3-dev tk-dev libgdbm-dev libc6-dev libbz2-
cd /usr/src
sudo wget https://www.python.org/ftp/python/3.7.7/Python-3.7.7.tgz
sudo tar xzf Python-3.7.7.tgz
cd Python-3.7.7
sudo ./configure --enable-optimizations
sudo make altinstall
```
Alternatively (much quicker): 
```
sudo add-apt-repository ppa:deadsnakes/ppa
sudo apt install python3.7
```
Make 3.7 the default with: 
`sudo update-alternatives --install /usr/bin/python python /usr/bin/python3.7 1`

openoms, 2020-06-28:
Also fails (on Odroid C4 - Ubuntu - python 3.8):
[image]
[image]
[image]
[image]  
Pasted the 24words
[image]
[image]
[image]

openoms, 2020-07-13:
For the record manual SDcard install looking ok with v1.6:
https://pastebin.com/raw/3BGKYHjV

will need to install `btrfs-progs` instead of `btrfs-utils`.

openoms, 2021-11-08:
My only C4 is dead :( .

closed 2021-11-08 by openoms