raspiblitz/raspiblitz #1200: Weird letters as style

Kapot, 2020-05-22:
Hey,

Not sure what I did wrong but my info screen looks like this (SSH on Kitty and on Termuis (Android).

[image]

Any way to fix this?

frennkie, 2020-05-23:
Did you check first https://github.com/rootzoll/raspiblitz/issues/353 and may be also https://github.com/rootzoll/raspiblitz/issues/314 ?

Kapot, 2020-05-25:
Hey, yes I did. But it seems to be in my pi rather then the software I use to ssh.

When I tried to change to HDMI output I got this message, maybe this helps?

	

> LANGUAGE = (unset),
> LC_ALL = (unset),
> LC_TIME = "nl_NL.UTF-8",
> LC_MONETARY = "nl_NL.UTF-8",
> LC_ADDRESS = "nl_NL.UTF-8",
> LC_TELEPHONE = "nl_NL.UTF-8",
> LC_NAME = "nl_NL.UTF-8",
> LC_MEASUREMENT = "nl_NL.UTF-8",
> LC_IDENTIFICATION = "nl_NL.UTF-8",
> LC_NUMERIC = "nl_NL.UTF-8",
> LC_PAPER = "nl_NL.UTF-8",
> LANG = "en_US.UTF-8"

frennkie, 2020-05-25:
Which image are you using?

Kapot, 2020-05-25:
v1.5-2020-05-08.img. I did raspi-config to setup timezone etc, maybe I made a mistake there? Anyone know how to change back to en_GB.UTF-8 for everything?

frennkie, 2020-05-25:
should be something like `sudo dpkg-reconfigure locales`

Kapot, 2020-05-25:
> Should be something like `sudo dpkg-reconfigure locales` (re-) install missing locales.
> 
> More info: https://www.thomas-krenn.com/en/wiki/Configure_Locales_in_Ubuntu

This seem to have fixed it. It seems normal now across devices!

rootzoll, 2020-05-27:
@frennkie anything we should add/change on the setup for next release?

frennkie, 2020-05-27:
Nope - not that I'm aware of...

Kapot, 2020-05-27:
The message of raspbian that the WiFi is disabled and you have to run raspi-config to set the country triggered me to do it. Can't disable this message I think?

rootzoll, 2020-05-27:
TODO: Check on the "WiFi is disabled" message for v1.6 release.

Kapot, 2020-05-28:
To complete it, this is the correct message shown:

> Wi-Fi is currently blocked by rfkill.
> Use raspi-config to set the country before use.

frennkie, 2020-05-29:
It briefly looked into it.. on Windows running `KiTTY` (a `PuTTY` fork) the setting (2) here is key:

[image]

frennkie, 2020-05-30:
> TODO: Check on the "WiFi is disabled" message for v1.6 release.

@rootzoll can you be more precise on what we want to check there? I have used `sudo raspi-config` 2-3 times today on a fresh v1.5 node (change Timezone, setup WiFi instead of LAN) and this works fine.

rootzoll, 2020-06-02:
@frennkie if you tested on windows and this was working fpr you, then we can close this issue.

closed 2020-06-02 by rootzoll