raspiblitz/raspiblitz #1205: Move BlitzTUI log to RAM disk

frennkie, 2020-05-23:
The RAM disk if a nice way to avoid reads and writes to either the SD or the external disk. Several things can be moved; the BlitzTUI log should be  a prime example.

hermanlim, 2020-05-28:
As of a couple of days ago, RPI4 supports booting from USB!
https://www.tomshardware.com/how-to/boot-raspberry-pi-4-usb

Will test it out and report back, this could be a good way to reduce wear on SD card (by eliminating it completely once initial setup completed) or also to simplify setup.

rootzoll, 2020-06-03:
@frennkie after a day of running the BlitzTUI with RAM logs it constantly restarts and the error in the logs is `No space left on device` on `/usr/lib/python3.7/logging/handlers.py`.

This is how the pi directory looks like:
[image]

frennkie, 2020-06-03:
Did you use the sync script? If yes... Was BlitzTUI updated to 0.48.1? I limited the size to 2 files with max 2 MB

rootzoll, 2020-06-03:
OK BlitzTUI was not updated correctly - I think there was a "sudo" missing to remove the old version. Now its updated and runs OK so far.

frennkie, 2020-06-03:
I started the ram disk initially 4mb...2x 2Mb log are already 4mb... I created a PR for increase to 8 MB (I guess we could go to like 32MB...

rootzoll, 2020-06-03:
32 seems like a good number :)

rootzoll, 2020-07-24:
Looks good. Closing issue.

PR #1227 (2020-06-02): Move tui log
fix #1205
closed 2020-07-24 by rootzoll