raspiblitz/raspiblitz #1217: Different time zones RTL

inpharmaticist, 2020-05-26:
I recently changed the time zone of my node (via sudo raspi-config) and RTL still shows old and new transactions in the original time.

openoms, 2020-05-26:
just restart the node

closed 2020-05-26 by openoms

inpharmaticist, 2020-05-26:
Restarted, made a new payment, confirmed date on the node. RTL still showing 7 hours ahead.

openoms, 2020-05-26:
Are you expecting RTL to change the times retrospectively in the lnd database?

The time of ab invoice creation for example.is matched with the time on the underlying OS.

Check the time on your node with the command:
`date`

inpharmaticist, 2020-05-27:
No, for each of my attempts above, I made new payments that also weren't correct. Date on node is currently correct.

[image]

FWIW times were corrected on Zeus. Seems like an RTL issue.