raspiblitz/raspiblitz #1218: RTL not paying invoices

inpharmaticist, 2020-05-26:
I've been trying to pay LN invoices (such as from yalls.org) and RTL doesn't pay them. Last week it was giving an error (don't remember the details). Now, it just doesn't do anything.  I can pay through the raspiblitz menu or the terminal commands, though.

openoms, 2020-05-26:
please provide more details.
Browser used, error messages etc.

Also seems like to be an RTL problem not raspiblitz

CoderBoogs, 2020-05-26:
@inpharmaticist i been trying to create a pay article on yalls and it never worked. Have you tried other lnd wallets besides RTL?

frennkie, 2020-05-26:
Same for me.. I can pay a yalls.org invoice using `lncli payinvoice ` and can browser RTL (e.g. balances and channels) but paying an invoice from RTL fails. 

There is an error in the log which I will have a look at.. check yourself: `sudo journalctl -f -u RTL`

frennkie, 2020-05-26:
The is an issue on RTL and there appears to be a fix in `RTL` version `0.7.1` https://github.com/Ride-The-Lightning/RTL/issues/283 "Firefox Payment Failure Issue"

frennkie, 2020-05-26:
Can config.. Tor Browser (Firefox) does not work. Chrome works.

openoms, 2020-05-26:
Try updating RTL with:

`sed -i s/v0.7.0/v0.7.1/g /home/admin/config.scripts/bonus.rtl.sh`

and reinstall from the menu.

frennkie, 2020-05-26:
Yes.. that fixed it for me: https://github.com/rootzoll/raspiblitz/commit/ca8f5dd39118ed94dc5e3f08a91e843587a1b2d8

inpharmaticist, 2020-05-26:
> please provide more details.
> Browser used, error messages etc.
> 
> Also seems like to be an RTL problem not raspiblitz

It was on Linux Mint using Firefox.  Followed:

>`sed -i s/v0.7.0/v0.7.1/g /home/admin/config.scripts/bonus.rtl.sh`
Removed and re added from "Activate/Deactivate Services" menu

Working now. Thanks.

closed 2020-05-26 by inpharmaticist

CoderBoogs, 2020-06-01:
Correction! I am having the same issue my apologies for providing mis information.

PR #1219 (2020-05-26): Update RTL to v0.7.1
Fix #1218
commit ca8f5dd3 (2020-05-26): Update RTL to v0.7.1 (#1219)

Fix #1218