raspiblitz/raspiblitz #1222: Monitor initial Blockchain Sync & detect Problems

rootzoll, 2020-05-27:
From time to time people report that the initial Bitcoin validation stalls - like this:
https://github.com/rootzoll/raspiblitz/issues/756#issuecomment-633209542

It would be great to monitor the sync and if progress stalls for a longer time to try counter measures like restarting bitcoind or system.

rootzoll, 2020-12-10:
like reported here #1859

rootzoll, 2020-12-12:
To get alternative seeds to kickstart in a 0 peer scenario - considering to use DNS seed.bitnodes.io

needed install: `sudo apt-get install -y dnsutils`
see: https://bitnodes.io/api/#dns-seeder

based on IPv4, IPv6 or Tor use the correct record

rootzoll, 2020-12-15:
Best to use the https API of bitnodes.io (over Tor).

rootzoll, 2020-12-20:
I am not able to call the bitnodes API `https://bitnodes.io/api/v1/snapshots/latest/` from Tor --> that triggers a CAPTCHA block by cloudflare what makes the API not usable. So the call is for now always clearnet .. what is okisch for now because the API call does not contain any node id that can tied to the IP ... BUT for v1.7 this should get a better solution:

I mailed the author asking for a .onion addreess. If that is not an option I would recommend a small proxy script that calls once a day `https://bitnodes.io/api/v1/snapshots/latest/` and offers a hidden service for raspiBlitz user to call anon.

NilByte, 2020-12-21:
Or use the bitcoin seeds list.

https://github.com/bitcoin/bitcoin/blob/cddcd22ab326428b461f1937f1db4d1329e46faf/contrib/seeds/README.md

rootzoll, 2020-12-21:
@NilByte the kick-start is for edge cases where bitcoind itself is not able to get started with peers (they use DNS seed lists internally) So using a non DNS method delivering a different peer list should be our best shot to circumvent such edge case situations.

we-do-it-lu, 2020-12-28:
Hello. Having the same issue and trying to follow-up with little experience. 
I ran `bitcoin-cli getnetworkinfo | grep "connections" | tr -cd '[[:digit:]]'` 
It returns `0`
I understand somehow my node is no longer connected to any nodes anymore (would _love_ to know the most common(s) causes of this is.
Then as suggested to get alternative seeds to kickstart in a 0 peer scenario I need to run:
`bitcoin-cli addnode "**IP_AND_PORT_OF_OTHER_NODE**" "onetry"`
It is stated to use DNS seed.bitnodes.io using utility tool 'dig' which is installed with the dnsutils by running the following command: `sudo apt-get install -y dnsutils` 
OK. dnsutils is installed.
I look at the link you shared https://bitnodes.io/api/#dns-seeder but don't know what to do with it... I ran the command: 
`dig TXT x5.seed.bitnodes.io` because my Raspiblitz is configured with Tor.
What do I do now? I run `bitcoin-cli addnode "**IP_AND_PORT_OF_OTHER_NODE**" "onetry"` but with what **IP_AND_PORT_OF_OTHER_NODE**?
Thank you in advance.

we-do-it-lu, 2020-12-29:
OK. This prevented me from sleeping, but got it resolved in the end:
As I realised that I have a 'prod' node that as been running for some time now I realised I could kickstart my blockchain synchronisation of 'test' node using 'prod' node.
I don't know why `bitcoin-cli addnode "**LOCAL_IP_OF_PROD_NODE**" "onetry"` was not working (returned immediatly doing nothing it appears) I had to use its Tor address: `bitcoin-cli addnode "**TOR_ADDRESS_OF_PROD_NODE.onion**" "onetry"` (then the command took a couple of seconds to return).
I am happy to say at the time I am finishing this post my 'test' node has finished synchronising the blockchain and look ready to play with :-)

rootzoll, 2021-01-12:
Testing looks good. Closing for v1.6.3 release.

closed 2021-01-12 by rootzoll